stawel/cheali-charger #215: Charging program for alkaline batteries

S-trace, 2017-03-04:
Hello.

Is it possible to add special program for charging disposable alkaline batteries, described in https://en.wikipedia.org/wiki/Recharging_alkaline_batteries ?
`Low-ripple direct current is not suitable for charging disposable alkaline batteries; more suitable is a current pulsed at a rate of 40 to 200 pulses per second, with an 80% duty cycle. Pulsed charging appears to reduce the risk of electrolyte leakage.`

I am now using charging program for two 1000mAh NI-MH cells (program for one cell is failing due to voltage limit), and it stops by deltaV (-317mV) after 5 hours and ~500mAh charged.

Thank you.

diggit, 2017-03-04:
Hi,
it is not possible with stock charger HW. These chargers have high output capacity preventing rapid changes. You'll have to remove it, but then normal charging would not be possible/safe/precise/reliable.

S-trace, 2017-03-04:
Okay, thank you.

closed 2017-03-04 by S-trace

specing, 2017-10-07:
Interesting.

Would it be possible to sink that excess capacity via the balance circuit?

stawel, 2017-12-07:
> Would it be possible to sink that excess capacity via the balance circuit?

hm.. interesting idea, it is probably possible to use the "discharge" circuit inside the charger for that,
but this is not something cheali-charge is designed for (we assume the charging current don't change rapidly)

specing, 2017-12-08:
But that assumption would only affect metering, no? Varying balancer sinks should not change (at least not by a lot) the  current through the DC-DC converter, but only current through the battery.