stawel/cheali-charger #216: Operation timer wraps to 0 on 65536'th second (1092,25 minutes)

S-trace, 2017-03-10:
Timer on display and UART timestamp wraps to 0 while long operations.
Here is part of UART log with issue:
```
$1;9;65530.9;4014;524;2874;210;1131;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;53
$1;9;65531.7;4014;524;2875;210;1131;3;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;58
$1;9;65532.5;4014;524;2875;210;1131;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;58
$1;9;65533.2;4014;524;2875;210;1131;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;60
$1;9;65534.0;4014;524;2875;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;58
$1;9;65534.8;4014;524;2875;210;1132;3;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;99;73;23840;61
$1;9;65535.6;4014;524;2875;210;1132;3;0;19056;4014;0;34;14;4;0;344;0;0;0;0;0;344;95;73;23840;61
$1;9;0.4;4014;524;2875;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;63
$1;9;1.2;4014;524;2875;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;56
$1;9;2.0;4014;524;2875;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;57
$1;9;2.8;4014;524;2876;210;1132;3;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;99;73;23840;61
$1;9;3.5;4014;524;2876;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;62
$1;9;4.3;4014;524;2876;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;63
$1;9;5.1;4014;524;2876;210;1132;2;0;19055;4014;0;34;14;4;0;344;0;0;0;0;0;344;97;73;23840;60
```
Maybe i met this bug because my Imax was powered on several days in a row, and uint32_t currentTime; was overflowed? 
0xFFFFFFFF (UINT32_MAX) / 1000 (milliseconds in second) / 86400 (seconds in day) == 49,71 (days of uptime until uint32_t currentTime will overflow), but mine Imax uptime is much less (maybe a week or so).

stawel, 2017-03-10:
hm...
generally the program code size suffers when I use 32bit integers, so I try to use them as little as possible, 
so I would say this bug should be fixed by changing:
(A)
https://github.com/stawel/cheali-charger/blob/master/src/core/drivers/SerialLog.cpp#L219
to printLong(currentTime/1000); 

>This issue looks to be because types mismatch:
https://github.com/stawel/cheali-charger/blob/master/src/core/drivers/debug.h#L33
inline void print(uint32_t x) { SerialLog::printUInt(x); }

I would consider this as a second bug (not related to SerialLog), should probably be:
(B)
inline void print(uint32_t x) { SerialLog::printLong(x); }
(generally debug.h is only for debug purposes, it's never used in release code)

> Maybe printUInt() should take uint32_t instead of uint16_t?

this is also an option, but in all the cheali-charger code Uint means a uint16_t (see lcdPrint.h),
so I would rather stick to it.

>There is same uint16_t overflow issue in lcdPrintTime():
https://github.com/stawel/cheali-charger/blob/master/src/core/drivers/LcdPrint.h#L67
https://github.com/stawel/cheali-charger/blob/master/src/core/drivers/LcdPrint.cpp#L163
void lcdPrintTime(uint16_t timeSec)

this is "intentionally" (not really, but it assures smaller program size), we only display 3 digits for minutes,
and 65536/60 = 1092, we could change the lcdPrintTime implementation but then 
we also need to change:
(C)
src/core/screens/ScreenCycle.cpp:52:    lcdPrintTime(cyclesHistoryTime[c*2]);
src/core/screens/ScreenMethods.cpp:120:    lcdPrintTime(Monitor::getTimeSec());
src/core/screens/ScreenMethods.cpp:123:    lcdPrintTime(Monitor::getTotalBalanceTimeSec());
src/core/screens/ScreenMethods.cpp:125:    lcdPrintTime(Monitor::getTotalChargeDischargeTimeSec());
(the Monitor::get... methods and cyclesHistoryTime[] )

I see you are quite familiar with the cheali-charger code,
maybe you could fix the (A) and (B) bugs and add a pull request?
unfortunately you would have to test it, and this will take time :)
(if you don't have the time, I will fix it this weekend)

regardless the (C) issue, I'm not sure if it's worth fixing, but If you think otherwise,
let me know, maybe some extra bytes won't hurt.

S-trace, 2017-03-10:
I'll try to fix it. 
Looking on https://github.com/stawel/cheali-charger/commit/c563f9fe2e4a7b33212e386962d87312aab0aa86 - there is your attempt to fix this issue (not merged in master for some reason, but that commit is useful). 
In particular, https://github.com/stawel/cheali-charger/commit/c563f9fe2e4a7b33212e386962d87312aab0aa86#diff-b2efa6c39fa0c1650dfb750cd4585a70L49 looks like pretty useful change.

> we could change the lcdPrintTime implementation
lcdPrintTime() could be changed to:
```
void lcdPrintTime(uint32_t timeSec)
{
	int16_t minutes=timeSec/60;
	if (minutes > 999) {
		lcdPrintUnsigned(minutes, 5, '0');
		lcdPrintChar('m');
	} else {
		lcdPrintUnsigned(minutes, 3, '0');
		lcdPrintChar(':');
		lcdPrintUnsigned(timeSec%60, 2, '0');
	}
}
```
This way we have original XXX:YY time format with second-precision time time for regular operations and XXXXXm time format with minute-precision time for very long operations. 
I'm think, ~49.7 days (0xFFFFFFFF/1000/60 == 71582 minutes) should be pretty enough for anybody.

stawel, 2017-03-10:
> Looking on c563f9f - there is your attempt to fix this issue (not merged in master for some reason, but that commit is useful).
In particular, c563f9f#diff-b2efa6c39fa0c1650dfb750cd4585a70L49 looks like pretty useful change.

hm.. I forgot about this commit... (probably not enough sleep),
ok so if it is in the "develop" branch then it will be merged
the reason why it's not done yet is that the "menu" code was heavily refactored 
and I need to do more tests, but everything should work. 

> lcdPrintTime() could be changed to: ....

looks good to me,
if you would like to add you "lcdPrintTime" code, then it should be in "develop" branch,
(take into account "int8_t dig") this way I will have less merge conflicts.

stawel, 2017-03-11:
ok, I've merged master and develop branches, hex files are available in 
https://github.com/stawel/cheali-charger/tree/master/hex/unstable

S-trace, 2017-03-15:
I have commited changes related to this issue to develop branch based on 2.01 release.
Could you take a look on it?

stawel, 2017-03-15:
looks great to me, commits merged (fast forward)