stawel/cheali-charger #217: Huge battery resistance dropout in one second - software bug or physical phenomenon?

S-trace, 2017-03-14:
Charging alkaline AAA cell using 2S 1000mAh NiMH charging program (at 100mA current) in original Imax B6 i noticed charging current increase and then huge battery resistance drop. 

Here is part of serial log showing the issue:
> $1;1;4752.9;2000;97;127;19;18;0;0;19066;0;0;0;0;0;0;528;0;0;0;0;0;23900;0;0;0;10
> $1;1;4753.7;2000;97;127;19;18;0;0;19066;0;0;0;0;0;0;528;0;0;0;0;0;23900;0;0;0;5
> $1;1;4754.4;2000;97;127;19;18;0;0;19066;0;0;0;0;0;0;528;0;0;0;0;0;23900;0;0;0;1
> $1;1;4755.2;2001;97;127;19;18;0;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;23900;0;0;0;6
> $1;1;4756.1;2070;141;127;29;18;0;0;19066;0;0;0;0;0;0;528;0;0;0;0;0;23920;0;3;446;63
> $1;1;4756.9;2086;144;127;30;18;0;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;23920;0;4;61248;58
> $1;1;4757.7;2093;145;127;30;18;1;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;23920;0;4;61248;49
> $1;1;4758.6;2095;144;127;30;18;0;0;19066;0;0;0;0;0;0;528;0;0;0;0;0;23920;0;4;61248;56
> $1;1;4759.3;2096;144;127;30;18;0;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;23920;0;4;61248;62
> $1;1;4760.2;2099;144;127;30;18;0;0;19065;0;0;1;0;0;0;528;0;0;0;0;0;23920;0;4;61248;57
> $1;1;4760.9;2099;144;127;30;18;0;0;19065;0;0;1;0;0;0;528;0;0;0;0;0;12940;0;4;61248;54
> $1;1;4761.8;2100;144;127;30;18;0;0;19065;0;0;1;0;0;0;528;0;0;0;0;0;12940;0;5;56514;60
> $1;1;4762.5;2100;144;127;30;18;0;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;12950;0;5;56514;48
> $1;1;4763.4;2100;144;127;30;18;0;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;12950;0;5;56514;48
> $1;1;4764.2;2101;144;127;30;18;0;0;19066;0;0;1;0;0;0;528;0;0;0;0;0;12950;0;5;56514;48

Is it software bug or physical phenomenon?
Why charger did increased charging current from 97 to 144mA?
Attached full charger log and visualized chart for it.
[alkaline_1.txt](https://github.com/stawel/cheali-charger/files/842297/alkaline_1.txt)
[image]

stawel, 2017-03-14:
could you get me your eeprom.bin from your charger (with the "program" you used)?
And which exactly firmware file are you using?

S-trace, 2017-03-15:
First log was captured with https://github.com/S-trace/cheali-charger/tree/master
Then i upgraded to https://github.com/S-trace/cheali-charger/tree/develop (based on release v2.01) but issue is still there.
Here is log and chart from https://github.com/S-trace/cheali-charger/tree/develop firmware and another battery:
[alkaline_2.txt](https://github.com/stawel/cheali-charger/files/843525/alkaline_2.txt)
[image]
Here is EEPROM read from charger running on 2.01 firmware:
[eeprom_issue_217.hex.txt](https://github.com/stawel/cheali-charger/files/843530/eeprom_issue_217.hex.txt)
I used 8th program (NiMH 1000mAh/2),

stawel, 2017-03-19:
Thanks for the scripts I willa add them to /utils/"something" :)

> Power in W/100 is a bit confusing, maybe it will be better to output pover value in mW, like other values?

it's not only Power where we have a 1/100 factor, see:
https://github.com/stawel/cheali-charger/blob/master/utils/cheali-logviewer/chealiparser.py#L14-L17

we use such a factor to be able to hold a reasonable value in a uint16_t type.

stawel, 2017-03-19:
ok, I finely found time to look into your eeprom.

Regardless the 97mA->144mA jump -
it looks like your current calibration is wrong.
Please calibrate you charging current again.
(I'm guessing during 50mA calibration you probably pressed "START" too soon,
before charging current stabilized - or you accidentally fast pressed "START" twice).

When it comes to the battery resistance, there is a bug in the firmware,
(resistance should not constantly change) - I'm still investigating.