stawel/cheali-charger #218: Make for Imax b8+ with 5A discharge

tobentobi, 2017-04-23:
Dear Pawel,
I use your great software since a few week on my b6 orig and on my b8+ orig as well. both are working fine instead of the  discharging on my b8+:
i flashed the HK EVO 8 150W FW and with this I cant use the 5A discharging of my B8+, because the ECO8 is limited by 1A.
Cant you compile your ECO8 project with the 5A (20W limit) dischare current for the  B8+? 

Best Regards, -tobentobi-

diggit, 2017-04-23:
Hi,
ECO8 has 5A current limit, problem is 5W power limit. Closest is 150W turnigy which has 25W discharge limit. I've compiled ECO8 with discharge limit modified to 20W for you.
[cheali_fw_ECO8_D20W.zip](https://github.com/stawel/cheali-charger/files/949539/cheali_fw_ECO8_D20W.zip)

tobentobi, 2017-04-24:
Only 5h to discharge a 6s5000 instead of 20...
Well done - thanks a lot!

diggit, 2017-07-04:
If this issue solved? If yes, close it please :)

closed 2017-07-08 by tobentobi

rn777, 2020-09-24:
Please add in cheali_fw_ECO8_D20W option power mode. Option Stabilisation at voltage. Thank's a lot.