stawel/cheali-charger #219: Two versions of Turnigy Accucell 6

list1999, 2017-04-27:
Here is a review of two versions of Turnigy Accucell 6 (www.microtforum.com/viewtopic.php?f=25&t=21751), I have the first version, completely similar to the imaxB6 original. What kind of firmware should I choose - cheali-charger-imaxB6-original_2.01-e10.3.12-20170311_atmega32.hex or cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex?

diggit, 2017-04-27:
Hi,
**cheali-charger-Turnigy-Accucel-6-50W-6A** is the right one. AFAIK version 2 has not compatible MCU with cheali-charger FW.

closed 2017-04-27 by list1999

fiveangle, 2017-04-27:
There are multiple versions of the Turnigy 6A.  If the board inside is labelled "SK6000A 10/10/2009" "PCB 6A" then the Turnigy hex above will not work and the imaxB6 hex you stated above will work, but will only get you 5A.  I have this board and have compiled for 6A (hex attached):

[imaxB6-original_6A_atmega32.hex.txt](https://github.com/stawel/cheali-charger/files/962330/imaxB6-original_6A_atmega32.hex.txt)

list1999, 2017-04-28:
I have this board (https://yadi.sk/i/IVz7PwWq3HTn3V), my only modification is to solder the connector for firmware. Recorded the firmware cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex. Calibration was done with the aid of a device Hold Peak HP890CN. Calibration is saved, Charging is working.

fiveangle, 2017-04-29:
That's the board. Good to hear the release 6A FW works. I don't know why calibration couldn't complete with the one I have. But haven't tried 2.01 so may check it out at some point (but charger working perfectly with the ImaxB6 target with max current increased to 6A.

diggit, 2017-04-29:
@fiveangle If it fails, post error code, so we can locate problem.