stawel/cheali-charger #221: Clarify list of supported chargers

aheyer, 2017-06-26:
Please add a comment to the Turnigy Accucell 6 50W 6A firmware that this charger is not identical to the Turnigy Accucell 6 50W 6A **with LiHV options**! The latter uses the unsupported Megawin MCU. I learned it the hard way: Lazily I hacked the cheali firmware for the Accucell 6 50W 6A into the Updater-EXE from HobbyKing. Not hard to guess what happened. I resurrected the charger [this way](https://github.com/stawel/cheali-charger/issues/173#issuecomment-310939391).

diggit, 2017-07-04:
In [Hardware section](https://github.com/stawel/cheali-charger#hardware):

  **WARNING**: You can get two chargers with the same name but different CPUs even if you buy from the same seller. Only chargers with a atmega32 or M0517LBN CPU are supported.

IMO, it is quite clear.

aheyer, 2017-07-11:
Do you think I'm stupid? That are two different chargers and not one with different hardware batches. Furthermore the LiHV version covers the CPU with the LCD module so you can't see if it's compatible to cheali-charger. Please show me one Turnigy Accucell 50W 6A LiHV with a supported CPU! If you can't my wish remains: Please update the build names of the Turnigy Accucell 50W 6A chargers to be incompatible with later revisions that have LiHV programs.

diggit, 2017-07-11:
> Do you think I'm stupid?

Never said that, I am just used to that people overlook some info easily. I do too...

>That are two different chargers and not one with different hardware batches. 

Just some additional FW features. Hardware is same. MCU varies. I can agree, that I haven't seen charger with LiHV and supported MCU ath the same time.

> Furthermore the LiHV version covers the CPU with the LCD module so you can't see if it's compatible to cheali-charger.

Most of 50W and 80W chargers have this layout. LCD is on pin headers, so it can be easily bent up to see MCU.

What are changes you suggest to do? Add note to Supported/Unsupported HW, that Turnigy Accucell with LiHV feature are most probably unsupported due to incompatible MCU?

aheyer, 2017-07-11:
> >Do you think I'm stupid?
> Never said that, I am just used to that people overlook some info easily. I do too...

Never mind! I expected that someone will point me to the documentation. Of course I read it before. But being lazy I didn't look at the hardware. Just comparing the label on my Turnigy with the offered BIN files. :-)

> Just some additional FW features. Hardware is same. MCU varies. I can agree, that I haven't seen charger with LiHV and supported MCU ath the same time.

Do you really think that only the MCU varies between LiHV and Non-LiHV 50W 6A Accucell 6? Of course the layout of my LiHV from 2016 looks similar but it's different to all the pictures about supported chargers inside the documentation. For example the ISP-Header is missing- or to be exact- it's placed on the other side in the middle of the board without labels.

From the documentation I thought that these warnings are for all the Imax B6 clones out there. (Is it really a clone with a different MCU with own firmware?) I supposed that every Turnigy version might have its own but always the same hardware. There's only one firmware update file available at HobbyKing what justifies my reasoning that there is only one LiHV hardware out there.

> LCD is on pin headers, so it can be easily bent up to see MCU.

I can't believe that! My LCD module is soldered on the pins and lies directly on the plastic spacers. IMHO there's no play to bent the module away. Or am I too scared?

> What are changes you suggest to do? Add note to Supported/Unsupported HW, that Turnigy Accucell with LiHV feature are most probably unsupported due to incompatible MCU?

Yes. And place an "Non-LiHV"-Tag on the firmware file for the [Turnigy Accucell 50 W 6A](https://github.com/stawel/cheali-charger/blob/master/hex/cheali-charger-Turnigy-Accucel-6-50W-6A_2.00-e10.3.12-20161016_atmega32.bin) (HEX file as well). Just to make sure that lazy people like me won't attempt to flash this firmware without opening the case. :-)

Noorudeen, 2017-08-16:
This method worked for my b6 mini, which i almost gave up on! thanks