stawel/cheali-charger #225: Is cheali-charger compatible with Mystery B6?

D4nko0, 2017-07-18:
I got this charger 8 years ago and it's still working great but its missing some useful functions like charging LiHV and measure internal resistance of a battery . 
Question is - that .hex i can upload to this board ?  Its an 50w b6 clone with Atmega32L cpu.
[image]
[image]
[image]

Kindest regards, Vlad.

diggit, 2017-07-18:
Hi @D4nko0,
your charger is matching B6 pinout at least for LCD and [layout looks very similar](https://raw.githubusercontent.com/stawel/cheali-charger/master/docs/imaxB6/imaxB6-original-front.jpg). What frequency is that ctrystal? (oval metal package)

@stawel did you try/support any ATmega charger with 8MHz crystals? This one has ATmega32**L** which has maximal freqency 8MHz according to datasheet.

D4nko0, 2017-07-18:
funny but crystal is 16mhz , just uploaded cheali-charger-imaxB6-original_2.00-e10.3.12-20160613_atmega32.hex , will try to calibrate .

stawel, 2017-07-18:
>@stawel did you try/support any ATmega charger with 8MHz crystals? This one has ATmega32L which has maximal freqency 8MHz according to datasheet.

hm.. a 8MHz crystal is currently not supported (but If it will be needed we can probably add it)
but I didn't see any charger with a 8MHz crystal yet, 
I did however see chargers with ATmega32L and a 16MHz crystal (this is probably the case)

[image]

D4nko0, 2017-07-18:
Update , for now all working fine , @stawel awesome work .
I will test it with my batteries and reply to you the results.
Thank you.

diggit, 2017-07-19:
@D4nko0 Enjoy new functionality :)

@stawel 
I tested ATmega8A on 24MHz and it worked ok, but I would not expect device running out of specs in commercial device.

> but If it will be needed we can probably add it

IMO that could be problem, half frequency means half resolution of PWM. Is such resolution enough to properly control SMPS and discharging load?

thliaw, 2019-11-06:
Hi guys, i know i have been late to this party, i too have a mystery B6, may i know where are the pins that i can connect in order to flash with the new firmware?

D4nko0, 2019-11-06:
> Hi guys, i know i have been late to this party, i too have a mystery B6, may i know where are the pins that i can connect in order to flash with the new firmware?

make a picture of your charger board , micro and crystal

thliaw, 2019-11-07:
[image]
This Mystery B6 charger comes with ATMEGA32L and a 16MHz crystal, I will try to attach few more pictures of the board.
[image]

D4nko0, 2019-11-08:
Its not necessary , look at this datasheet of atmega32 [http://ww1.microchip.com/downloads/en/devicedoc/doc2503.pdf](url) first 6 pins are for programming.  That programmer do you have ?

thliaw, 2019-11-11:
> 
> 
> Its not necessary , look at this datasheet of atmega32 [http://ww1.microchip.com/downloads/en/devicedoc/doc2503.pdf](url) first 6 pins are for programming. That programmer do you have ?

Thank you for providing the datasheet of atmega32 mate, i will try to solder these  pins (Ground, Reset, MOSI, MISO, SCK) based on photos i found from the forum, i have a USB ASP programmer, may i know how should i proceed next? 
[image]
[image]

[image]

D4nko0, 2019-11-11:
Read this [https://github.com/stawel/cheali-charger/blob/master/docs/flashing.md](url)

This is HEX to upload for your charger [https://github.com/stawel/cheali-charger/blob/master/hex/cheali-charger-imaxB6-original_2.00-e10.3.12-20160613_atmega32.hex](url)

Neo59, 2020-05-19:
Hello there. Have you uploaded? Has the device worked?