stawel/cheali-charger #227: GT Power A8

alvesjc, 2017-08-25:
Hello.

Will the GT Power A6-10 firmware work on the GT Power A8 ?

I can I confirm this without trying to flash it ?

Help appreciated.

stawel, 2017-08-26:
according to this:
https://groups.google.com/forum/#!searchin/cheali-charger/GT$20Power$20A8%7Csort:relevance/cheali-charger/CaRsXnwvJdY/_5E3WQJpDgAJ

it should work with Turnigy Accucel 8 150W 7A firmware,
assuming your charger has a atmega32 CPU.

alvesjc, 2017-08-27:
Hi Stawel,

Ok, thank you for your input.

I'm noticing a huge diference in real battery voltage and the measured voltage in the charger, especially when discharging. The charger is terminating the discharge process very very early because of this.

The buit in calibration, only change a few milivolts, not enough.

This firmware has no limit in the calibration process?

Thanks

diggit, 2017-08-27:
Hi @alvesjc,
>I'll also see if it's possible to mod the charger and bring the voltage probs close to the battery independtly of the charging leads. This way the high current flowing in the cables and causing variable voltage drops would not be an issue.

IMO this is not necessary. Nixx cells rely on delta peak which is  difference over time and absolute value does not matter so much. Pb cells are quite tolerant. Lixx pack voltage is sum of all balancer inputs and voltage measured on charging terminals does not matter.

Mod should be possible, look into schematics in Troubleshooting section of README. BATT+ and BATT- are terminals you are looking for.

alvesjc, 2017-08-27:
Hi Diggit,
> Lixx pack voltage is sum of all balancer inputs and voltage measured on charging terminals does not matter.

This is what i'm seeing with the stock firmware, it is measuring the charge cable voltage, not the balance ports.

When you state this, you mean the Stawel firmware?

Thanks

alvesjc, 2017-08-27:
Hi again, you may see the diferences here:

https://goo.gl/photos/gK3tt4JzzgZzjrDX7

when the discharge end, the balancing port was stating 3.47V yet, while in main screen was stating 3V. 

It ended the discharge because it has reached 3V, and only discharged 1Amp/hr, when the battery still have a lot more to give.

diggit, 2017-08-27:
Yeah, I meant in cheali charger FW. 

Difference in voltage measured on charging leads and balancer is expected. However it should almost vanish with decreasing current during discharging. (if it is caused by wire resistance). At least in cheali, there is adaptive discharging, which lowers load current when getting closer to final voltage. (inverse behavior to CV/CC charging)

alvesjc, 2017-08-28:
Hi again.

Well, indeed, no need to mod anything, the firmware allows a correct tune.

Now I'll test some LG 18650 3AH batteries to access it against the datasheet.

The diferences was huge at start, but now all match my multimeter.

I've taken pictures of the charger dismantled, If you feel they are needed for the documentation, I'm very happy to contribute to your work.

I've added a 10pin header to the charger PCB, from now on it will be easy to program it.

Thank you very much for your work.

Regards,

diggit, 2017-08-29:
>The diferences was huge at start, but now all match my multimeter.

Charger flashed with cheali must be fully  (re)calibrated. Default calibration is rough. I suggest to calibrate voltage with fully charged batteries (near 4.2V per cell).

You can post pictures right here for future reference, all contributions are welcome.

alvesjc, 2017-08-29:
Hi.

See the pictures below.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Regards,

alvesjc, 2017-08-29:
The pinout in this header acordingly to the printed numbers in the PCB is:
0 - 5V
1 - MISO
2 - SCK
3 - RST
4 - NC
5 - MOSI
6 - 9 - GND

alvesjc, 2017-08-29:
I forgot to see the discharge fet reference, anyone knows it's reference?

I'm planning on increase the discharge capacity well beyond the 25W mark...

Help appreciated.

diggit, 2017-08-29:
Thanks for photos and pinout.

Discharge power is limited by charger capability to dissipate such amount of power (and still maintain sane temperature) rather than transistor itself. If you want to increase discharging current, you need current sensing resistor with lower value.

There are some discharge power mods documented on chealis' google group.

alvesjc, 2017-08-29:
Hi,

You're welcome.

Yes, right now the fet + current sense resistors are the only ones dissipating power, at least is what i'm perceiving from the schematics.

My ideia is to add a external series resistor between Battery + and the charger.

This way you get a significant lower voltage across the FET, thus, less power dissipated in the charger case.

The limit would be only the FET maximum current. It's almost guarantee that it can handle 20 to 30A, but I want to make sure.

For my calculations,

to discharge a 8S Lipo at 20A I would need a more or less 1.5ohm external resistor with 600W, very feasible with aluminium heatsink resistors (6x 1ohm 100W resistors, 3 series paralell with other 3 series).

Playing with the resistors arrangement, it's possible to discharge lower voltages at same current.

Regarding the current sense resistor, yes, 4x 0.1 ohm 5 or 10W would be good to go I think.

I'll search those mods.

Thanks

diggit, 2017-08-30:
> Yes, right now the fet + current sense resistors are the only ones dissipating power, at least is what i'm perceiving from the schematics.

Yes, all power should be dissipated on transistor. Power loss on resistor is just side effect of its resistance. You need decrease its resistance to measure current in wider range (ADC has limited voltage range).

Disadvantage of external resistance is need for adjustment for each different cell count. You have to calculate it to have most of voltage drop on resistors and only small drop on internal transistor.

Hi @stawel, just quick question... During Lixx discharging is power loss calculated from current and voltage on power terminals or from voltage sum on balancer?
In case of power terminals, there should be no problem adding external resistance in series to offload some dissipated power to it. (and still no need for rising discharge power limits...) While correct battery voltage remains measured by balancer. Am I right?

stawel, 2017-08-30:
@alvesjc pics added to [docs/GTPowerA8/](../tree/master/docs/GTPowerA8)

@diggit when balance port is connected then it is the cell voltage sum, otherwise it is the terminals voltage.

> In case of power terminals, there should be no problem adding external resistance in series to offload some dissipated power to it.(and still no need for rising discharge power limits...) While correct battery voltage remains measured by balancer. Am I right?

so currently it's a problem (probably easy fixable), 
but there are also other problems:
1. when "adaptive discharge" is enabled we will gradually decrees the discharge current at end of discharge, this would spread the discharged  power between the connected resistors and the discharge transistor (at some point the power dissipation will exceed the FET limits)  
2. something similar will probably also happen when we start discharging, since we start at a relative small current ("Id"/10) and then skip to "Id" after a while.

The solution would be to add another discharge mode where the FET is turned on/off like a switch,
(I had plans to add this mode)
 but currently I'm not sure if it's worth it, since (as your mentioned) the resistors need to be different 
for each discharge current/voltage - it's just simpler to add a bigger FET with a heatsink (it's probably also cheaper)

alvesjc, 2017-08-30:
@stawel 

" it's just simpler to add a bigger FET with a heatsink (it's probably also cheaper)"

yes, indeed it's other possibility, route the fet cables out of the charger box and place the fet externaly or just build a new box for the charger.

" The solution would be to add another discharge mode where the FET is turned on/off like a switch,
(I had plans to add this mode)
"

But for this, you would need to remove or bypass the low pass filter in the current charger circuit, or insert a new FET in parallel to the existing one but controlled directly by the port PD5 of the ATMEGA32 with some additional logic gate to enable it or not.

This would need to redesign the charger circuit.

That's why I focused in the external resistors.

I'm avoiding to open the discharger again, but if no one knows it's reference I will open it again.

Knowing the FET, maybe we can come to a balance point with external resistors, allowing only one or 2 values for external resistors, for example 0.12ohm for a range of discharge voltage and current, 1ohm for other range. This is just an example.

Anyway, just storming ideias...
But I'll try to reach at at least 20A discharge.

I've also found the circuit diagram for the accuel 8150, I think it would be nice to have it in GTPower A8 Docs section.

[8150 full mod2.pdf](https://github.com/stawel/cheali-charger/files/1265270/8150.full.mod2.pdf)

Regards,

stawel, 2017-09-02:
>~~when balance port is connected then it is the cell voltage sum, otherwise it is the terminals voltage.~~

@diggit  my mistake, this was already fixed, It's just like you said, but the two additional issues remain.

@alvesjc
> This would need to redesign the charger circuit.

 not really, the FET can be turned on/off without changing the charger circuit, 
we just need to set the discharge current to MAX_DISCHARGE_I (or 0A to turn it off)
as long as the discharging current is less than MAX_DISCHARGE_I the FET should act as a switch.