stawel/cheali-charger #229: Unable to reach max Pc with 6S on imax B6

diggit, 2017-09-09:
Hi @stawel,
There is [thread on google goups](https://groups.google.com/forum/#!topic/cheali-charger/W5WednkXPqE) about problem in title. There is probably something hitting its' limit, but not sure what.

I have compiled upstream version and performed eeprom reset. My modified imag B6 is not able to reach maximal charging power with 6S. Testing is performed on discharged 1.3Ah LiPo battery, Ic set to 3A.

During charging:
- voltage is rising
- current is falling
- power is falling

Do you have any suggestion how to debug this? I am suspecting, that SMPS is hitting some internal limit.

silenec, 2017-09-10:
Hi, I found this:
https://github.com/stawel/cheali-charger/issues/206
https://groups.google.com/forum/#!topic/cheali-charger/1VULY0Jlh74

"the output voltage (battery voltage) ratio to input voltage (power supply voltage) must be smaller than 2 during charge
(or to be exact: 
         Vout/Vin < 1/(1-D) 
          where D = MAX_PID_MV_FACTOR - 1 
)
this is a safety "feature"  and probably should be re-thought.
A temporary workaround is to:
a) use a power supply with higher voltage
or
b) increase MAX_PID_MV_FACTOR to 1.60
 in: https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/generic/50W/SMPS_PID.h#L12"

Could this be it? My output to input voltage ratio is more than 2. 
Should I try to increase MAX_PID_MV_FACTOR? It seems like it is currently set to 1.5 ?

silenec, 2017-09-10:
Vout/Vin < 1/(1-D)   where D = MAX_PID_MV_FACTOR - 1 

When input voltage is 10V, charging voltage 25,2V and MAX_PID_MV_FACTOR is 1,5:
25,2V/10V < 1/(1-0,5)
2,52 > 2 = limited

When MAX_PID_MV_FACTOR is 1,61:
25,2V/10V < 1/(1-0,61)
2,52 < 2,56 = this should work, right?

stawel, 2017-09-10:
>Could this be it?

Yes, this is the limit you are hitting 

>Should I try to increase MAX_PID_MV_FACTOR?
>When MAX_PID_MV_FACTOR is 1,61: ...
>2,52 < 2,56 = this should work, right?

you could try to increase MAX_PID_MV_FACTOR gradually from 1.50, 1.51, ... to: 1.61
(going one step further only when charger works correctly at maximum possible output power)

unfortunately I don't remember why I've set MAX_PID_MV_FACTOR to 1.50
(it clearly should be set to MAX_PID_MV_FACTOR 1.564 for Imax B6 charger (minimum Vin is 11V)),
but I probably had a reason.

silenec, 2017-09-10:
Right, well I raised MAX_PID_MV_FACTOR to 1.61, compiled and flashed the firmware, reset eeprom, calibrated the charger and gradually raised the maximum watt limit in settings menu... everything seems fine now at 50W. I have multiple fans blowing over my chargers so temperature should not be an issue. My input voltage is actually almost 11V so my "real" MAX_PID_MV_FACTOR is somewhere around 1.58 which is ~1% increase from the 1.564 value. Hopefully it will be fine. I guess the default value of 1.5 could be raised at least a bit.

stawel, 2017-09-10:
>  I guess the default value of 1.5 could be raised at least a bit.

I could change this (even to 1.63) but this value should  probably depend
on the input voltage (when power supply voltage is higher this limit should be lower)
if I find time next weekend I will try to implement this 
(please ping me if I forget about it :/ )

silenec, 2017-09-10:
I would imagine that with higher input voltage the charger would still get limited by maximum power (50W or whatever the limit is for a specific target) and then the MAX_PID_MV_FACTOR value would not really do much because it is just maximum limit and the charger only uses what it needs to reach the watt power limit, no?
Or is there any difference if you charge at 50W with the same input voltage and MAX_PID_MV_FACTOR set to 1.6 and let's say 1.8?

stawel, 2017-09-10:
MAX_PID_MV_FACTOR set to 1.50 (1.63) only limits the maximum duty cycle to 50% (63%) in boost mode 
(max. duty = MAX_PID_MV_FACTOR - 1.)
see "continuous mode" in: https://en.wikipedia.org/wiki/Boost_converter
so it actually only limits maximum possible output voltage (not output power - but these two are correlated),
This limit has two purposes:
1. when charger was not properly calibrated, it still has a hard coded limit.
2. when PID overshoots, output voltage should not rise above Vo = Vi/(1-D) [where D is duty].
   (in boost "continuous mode" of course)

The firmware has  many other  software limits: max. output voltage, max. output power, max. output current, min. input voltage ... and so on, but those work only when the calibration was dane correctly (witch is not always the case)

> the MAX_PID_MV_FACTOR value would not really do much because it is just maximum limit and the charger only uses what it needs to reach the watt power limit, no?

assuming the charger is well calibrated AND I didn't do any stupid mistakes in the code ;)
then yes 

> Or is there any difference if you charge at 50W with the same input voltage and MAX_PID_MV_FACTOR set to 1.6 and let's say 1.8?

no there shouldn't be any differences, you should never reach this limit.

But if you are using a power supply with higher output voltage and your charger
is wrong calibrated (measured output voltage is lower than in reality)
then this limit is your last hope.

silenec, 2017-09-10:
That makes sense. Thanks!

closed 2018-08-26 by diggit