stawel/cheali-charger #230: B6 clone settings max Id

diggit, 2017-09-13:
Hi @stawel ,
I browsed though sources and found something weird. Why is **B6 clone** SETTINGS_MAX_DISCHARGE_I set to 0.3A and MAX_DISCHARGE_I to 1A? (both are m32)

Seems like some forgotten value from testing.

**B6 original** has not SETTINGS_MAX_DISCHARGE_I defined so it defaults to MAX_DISCHARGE_I.

[HardwareConfig.h](https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/imaxB6-clone/HardwareConfig.h#L32)

stawel, 2017-09-13:
It's because I have two imax B6 chargers:
1. one "original" - at least hobbyking says so, it's maximum discharge current is 1A
2. one "clone" (copy) - also hobbyking says so, it's maximum discharge current is 0.3A 
(this are hardware limits, resistors on current measurement circuit are set so)

so by default (eeprom reset) one has "max Id" set to 1A and the other to 0.3A,
of course the "max Id" setting can be increased up to 1A (MAX_DISCHARGE_I) in both cases
(just in case someone has a different "clone"),
but in my "clone" case, if I set "max Id" to 1A I get a "calibration error".

diggit, 2017-09-13:
Oh, that makes sense. I've never seen charger wit Id<1A. Thanks for clarification.

closed 2017-09-13 by diggit