stawel/cheali-charger #233: Minimum (dis)charging current

specing, 2017-09-23:
I was wondering how low I could go on my imax b6 and found some help in docs/calibration_error_codes.md. Within are several non-working wolfram alpha programs to help determine this:
e.g. the first one is {Fit {{702,100}, {10944,1000}},{1,x},x  , x 1}. Looking at Fit documentation at https://reference.wolfram.com/language/ref/Fit.html I can see that Fit[] takes 3 parameters, but there are 4 on the link. Additionally the first { should probably be after Fit and changed to a [ along with the last } changed to ]. There is also the stray? "1" in the vars field.

Anyway, even after changing all this what one gets out of this least-square-fit is a line and not a value that one could then use in max[] with the result of the other query (which too suffers from the above problems).

How does one determine minimum currents?

diggit, 2017-09-23:
Proper formulation is [`Fit[{{702, 100}, {10944, 1000}}, {1, x}, x]`](https://www.wolframalpha.com/input/?i=Fit%5B%7B%7B702,+100%7D,+%7B10944,+1000%7D%7D,+%7B1,+x%7D,+x%5D)

{1,x} means, that you wan to find coefficients for first order polynomial. (y = a\***1** + b\***x**, you want to find _a_ and _b_, btw first order polynomial is line equation...)

_x_ is adc value
_y_ is current (mA)

with numbers you provided, result is: `38.3128 + 0.0878735*x`, if you substitute 0 as ADC value (x) you get 38.3128mA as minimal current. (ADC number can't go to negative values)

Such current flows when ADC measures 0. Cheali FW will not allow lower currents to be set, because there is no way how to measure them.

specing, 2017-09-24:
The question was not what this formulation represents but how such a formulation was constructed in the first place. From what was said I gather that I should be looking at the minimum current that still results in a positive ADC value and the minimum current that can be generated with a positive driver?(I suppose this represents the on time of a mosfet) value. And then take the maximum of that (i.e. both values must be positive).

With this in mind that 100 should be changed to 50 (as the imax b6 calibration is done with 50mA and 1A currents). Now I get  0.092755*x -15.114236 and negative currents when driving with 0. Hmm, something is probably off. I guess I have to redo calibration anyway to get ADC values.

specing, 2017-10-07:
50ma: drive 645, adc: 645 (+- 2)
1A: drive 10915, adc: 10915 (+- 10)

value = 10.8105*current + 104.4737
At 1mA both pwm? and adc would read 115. Apparently there is no lower limit to the current? Is this normal for an atmega32 IMAX B6?

stawel, 2017-12-07:
> Within are several non-working wolfram alpha programs

hm.. links work quite well for me:
`{Fit[{{702,100}, {10900,1000}},{1,x},x] , x=1}`
return:
`38.1349`
(I'm assuming your browser changes somehow the links, TODO: add full formula to the text)

> how such a formulation was constructed

It's actually no the minimum (maximum) current your charger is able to deliver but rather 
some limits beyond which we display "Calibration error".
values below which a linear extrapolation of the calibration points would return a **x** below 1 or above 65400,
(would not fit in a uint16_t) 
(TODO: clarify this in text)

> With this in mind that 100 should be changed to 50

the 50mA Calibration point should rather be changed to 100mA (in most chargers this is the case)
(you can also do this in you charger). The ADC reading near 100mA is much more "linear" 
the near 50mA.
(TODO: change default calibration point to 100mA in all chargers)

> At 1mA both pwm? and adc would read 115. Apparently there is no lower limit to the current? Is this normal for an atmega32 IMAX B6?

that only means our extrapolation doesn't go below 1, but the output current doesn't behave linear near 
the ends,so you shouldn't go below 50mA.
(TODO: add to text)

specing, 2017-12-08:
The range of uint16_t is [0, 65535], not [1, 65400]. Ok, 0 would turn the thing off, but I don't see why 65400- 65535 are bad?

I see, will change it _one day_.

I'm actually interested in this for the purposes of charging alkaleak button cells, where going as low as possible is benefitial (until FDK starts producing LR44 eneloops ;D). Guess could work around it with balancer as a variable sink.

fiveangle, 2018-01-09:
@specing - if we wanted to monitor your progress to charge button cells, where would we go ?  Some branch here ? : https://github.com/specing/cheali-charger/

specing, 2018-01-10:
I have not started any work on alkaleaks. As pulsed charging appears to be the best method, we will probably have to utilize the balancer and therefore we could also sink excess current on it.

stawel, 2018-01-11:
> The range of uint16_t is [0, 65535], not [1, 65400]. Ok, 0 would turn the thing off, but I don't see why 65400- 65535 are bad?

The ADC has  10bit resolution,  maximum value is 1023 and it's shifted left 6 bits:
1023<<6 =   65472 (that's close to 65400 with a small margin needed for error detection)

> I'm actually interested in this for the purposes of charging alkaleak button cells (and because I can), where going as low as possible is benefitial (until FDK starts producing LR44 eneloops ;D). 

I'm not sure but the charger is probably capable of charging with very low current, unfortunately
the current measurement in ranges (approximately) 0mA-50mA  is not linear (see schematics, a Rail-to-Rail op-amp would probably help) , and we assume it is linear,
that is why we don't use this range. But with more advanced calibration you probably could use it.

specing, 2018-01-12:
I see. Anyway I think it would be possible to sink excess current via the balancer.