stawel/cheali-charger #234: HK ECO-6 50W compatiblity

NiklasVoigt, 2017-10-04:
Hi, Is that charger compatible?

**HK ECO-6**
Power: 50W
Charge: 5A
Discharge: 1A
Chip: Atmega32A
ISP: Same occupancy as in the firmware manual

PCB:
[image]

SMoskalev, 2017-10-04:
I've got same charger, but with a little bit different PCB. Fully compatible. Successfully flashed with common firmware.

NiklasVoigt, 2017-10-04:
Which firmware did you flashed ?

SMoskalev, 2017-10-04:
imaxB6-original-*

NiklasVoigt, 2017-11-09:
Works like a charm!

-Closed

Casetero, 2017-11-21:
Hello,

I have a Thunder AC680 that is aparently same pcb as yours. Mine has usb conection (shared with temp sensor), it uses the three PIN´s of the left of your PCB in your photo for the USB/Temp Sensor. Have you flashed the firm with the USB conection or going to the PCB direcltly (as shown in the description of the main cheali post)

NiklasVoigt, 2017-11-21:
The HK Eco6 shares a USB Port too.
Look here: http://transistor-man.com/echo-6.html

I've flashed through the PCB directly as shown in the Pictures of the Manual. Before that i testet everyone Pin with my Multimeter to be in the Safe side. But all Pins are the Same an i flashed with an usbasp the hex of the imaxb6 original

Casetero, 2017-11-21:
Thank you!
Do u think it could be flashed via USB?

NiklasVoigt, 2017-11-21:
I dont think that you can flash through the USB port

gpomykala, 2019-03-14:
Hi, does anyone know pinout on this PCB? It's HK ECO6 purchased around 2011
[image]

diggit, 2019-03-15:
Hi @gpomykala,
I'd expect something like [this one](https://github.com/stawel/cheali-charger/blob/master/docs/imaxB6/imaxB6-original-front.jpg).
Just verify, that second pin from right (bottom on photo) is connected to ground (should be largest polygon on rear side of the board).

gpomykala, 2019-03-15:
Thanks a lot. It did the trick:)

chrisgh1, 2019-09-03:
@gpomykala or anyone else that has the same PCB.  I'm trying to figure out the value of what I think is a resistor in the lower left corner of the PCB.  From the picture it only looks like it has a black and green band so not able to calculate the resistance.   I need to replace this as mine burned up in a short.  Any help would be appreciated.