stawel/cheali-charger #239: Flashing via USB in thunder AC680

Casetero, 2017-11-21:
Hello
I own a Thunder AC680. Here is a link :

https://www.amazon.com/Thunder-AC680-Professional-Dual-Power-Discharger/dp/8499000037

This charger has an USB port, and it comes with a UART to USB card to connect it to the computer for monitoring during the charge.

Can i use this port + card to flash the cheali charger to it?

Thank you.

diggit, 2017-11-25:
At first you have to find out what MCU does that charger use. If it is supported by cheali (AVR or Nuvoton ARM) then you can think about flashing cheali. Even if it is supported, I don't think, that flashing via UART is possible (at least with factory config and FW).

stawel, 2017-12-07:
if this charger is supported by cheali-charger then it is not possible,
all supported chargers (for now) have only one directional communication:  
charger -> PC   (from charger to PC)