stawel/cheali-charger #241: Support for NUC029LAN

iLLiac4, 2017-12-05:
Hi. Today i ghave received imax b6 charger with nuvoton NUC029LAN chip. Is this one supported.
There also seem to be serial port located on this board.
Here are the pictures of the chip and board:
https://www.dropbox.com/s/81u1nllygtn7des/20171205_191059.jpg?dl=0 
https://www.dropbox.com/s/rdaqme6wfllrjfo/20171205_194212.jpg?dl=0
https://www.dropbox.com/s/j1bfr9zcif31xaw/20171205_194225.jpg?dl=0

Stormwing01, 2018-02-13:
I have that exact one... If this software is not supported, where can I find one that is?

pampyras, 2019-10-28:
I have the same one, made it almost working

SenasLapinas-LT, 2021-01-03:
Can you share your progress?

pampyras, 2021-01-03:
@SenasLapinas-LT cannot make voltage regulator to work, so stopped for now, until have more time

SenasLapinas-LT, 2021-01-06:
@pampyras Can you share your findings so far? Do you able to find schematic of this board?

pampyras, 2021-01-06:
@SenasLapinas-LT contact me at gmail email, same like nickname,, will be simplier (galim ir LT)

pampyras, 2021-06-05:
@shadowlab if you need any tracked pinouts, I can share what I have

shadowlab, 2021-06-05:
> @shadowlab if you need any tracked pinouts, I can share what I have

That'd be great! I also crossposted to the google group at the following: 

https://groups.google.com/g/cheali-charger/c/hzF2EVrXjpo 

I'm still mostly poking around at things, and can't really test any physical device stuff until the programmer arrives. Any help is much appreciated.

pampyras, 2021-06-07:
Here you go https://github.com/pampyras/imaxb6_nuvoton/blob/main/src/hardware/nuvoton-NUC029/generic/50W/imaxB6-pins.h
But as I see my board is blue and has different layout, so I do not know if the pinouts will match.

shadowlab, 2021-06-08:
> Here you go https://github.com/pampyras/imaxb6_nuvoton/blob/main/src/hardware/nuvoton-NUC029/generic/50W/imaxB6-pins.h
> But as I see my board is blue and has different layout, so I do not know if the pinouts will match.

Thanks for providing that. I went through and there were a few changes with the pins on mine. I'll probably mess around with it when I have some time.

sur5r, 2021-09-25:
I received a "SkyRC Imax B6 Pro Evo" yesterday, genuine SkyRC product, which is also using the NUC029LAN. It reports itself as "HW: 2.0 FW: 1.06". I can provide PCB shots if wanted.

In contrast to the other pictures provided here, it does have a USB port. As the NUC029LAN does not have a USB peripheral, I guess SkyRC has implemented this via bit banging.

pampyras, 2022-04-23:
Hi, maybe someone made some progress on this MCU?

Akela77, 2023-01-22:
Please say what procent of progress with this microcontroller NUC029LAN? Where can i find finished firmware file?

joshgarde, 2023-01-22:
We have "alpha" support for the NUC029LAN; it mostly works, but there's still an ADC issue that causes the charger to believe there's a battery disconnect after a longer cycle. The .hex is not uploaded as I don't consider it to be in a stable enough state for release

https://github.com/stawel/cheali-charger/pull/316

diggit, 2023-01-22:
Is there some reliable source of chargers with NUC029LAN MCU? I'd like to have a look on issue, @joshgarde mentioned.

joshgarde, 2023-01-22:
The EVPeak CQ3 line (I think they rebranded recently) have been updated to use the NUC029LAN. The issue is that they're pricier chargers (~$180 on Aliexpress) because they're quad channel. I haven't looked into EVPeak's other offerings

joshgarde, 2023-01-22:
Programming the CQ3 line also isn't hard as the button header acts as a programming header as well. Standard ARM toolchain flashing works, but the MCU has to he unlocked first

chriscloyd006, 2024-12-04:
Hi can anybody help me ,How to copy and re write the IC NUC029LAN