stawel/cheali-charger #247: support for imax b6 60w

dwc56, 2018-01-05:
i put the firmware for imax b6 clone on my charger and it work just fine after calibrating it but the charger i put it on is a 60w charger with charge current max of 6amps the firmware only goes up to 5a is this something that can be changed?

stawel, 2018-01-05:
I'm assuming it's a atmega32 one?
could you find a link to chargers specs?

dwc56, 2018-01-05:
yes it is atmega32
and with the old firmware it could do 2a discharge
http://www.headsuphobby.com/SkyRC-iMax-B6-Mini-60-Watt-Balance-Charger_p_1379.html
http://www.skyrc.com/index.php?route=product/product/download&download_id=30

stawel, 2018-01-05:
please see:
https://github.com/stawel/cheali-charger/tree/develop/hex/unstable
search for "imaxB6-mini"

I hope this helps
Best regards
Paweł

dwc56, 2018-01-05:
Thank you for doing that

stawel, 2018-01-05:
please notice that you now have 2A discharge current but only 5W discharge power (based on spec)
so you only get 2A up to 5W/2A=2.5V output voltage.
the same is true for charging:
60W/6A = 10V (maximum output voltage for 6A output current)

dwc56, 2018-01-06:
i put that on my charger and after calibrating it if the max ic in settings is higher then 4.5a it errors 
 calib. error I charge 14

stawel, 2018-01-06:
hm... I would try to recalibrate the charging current again, 
the "imaxB6-mini" image is just a "imaxB6-clone" copy with limits set to 6A, 2A, 60W,
so if you previously had a bigger charging current then you calibration differs.

You should also change the calibration points "50mA" and "1000mA" to something higher 
maybe "200mA" and "2000mA", this way you calibration will be more accurate on higher current.

dwc56, 2018-01-06:
i changed that and re calibrated it and it has the same error

stawel, 2018-01-06:
hm... it's looks like your charger is just 4.5A, did you have a bigger charging current with the 
imaxB6-clone firmware?

dwc56, 2018-01-06:
yes with that it could go up to 5A and with the firmware that came on the charger it could charge at 6A

dwc56, 2018-01-06:
and with both firmware the discharging calibration is way off 300ma on the charger is about 1.2A if that matters but that can be fixed in the calibration

stawel, 2018-01-06:
hm... ok so maybe I have a bug in the unstable version,
could you so something for me?
1. flash your charger with the imaxB6-mini firmware:
   a) calibrate your charger, and download the eeprom (after calibration)
2. flash your charger with the imaxB6-clone firmware:
   a) calibrate your charger so that you have the 5A charging current, and download the eeprom (after calibration)

and send them to me, I would like to compare the two eeproms.

It is possible that you don't have to recalibrate after flashing, in this case check if you can charge up to 5A and send me the eeprom's (two).

dwc56, 2018-01-07:
i did not calibrate i discharge on the 50w firmware  
50w [https://drive.google.com/open?id=1K02R4RA6RJPUsYK9PSCdOYNk92vUB15V](url)
60w [https://drive.google.com/open?id=1LEA-Lc23I4OTl_4MIvlQqnCpQgyZT_wQ](url)

dwc56, 2018-01-07:
the link does not work but if you paste it browser it works

stawel, 2018-01-07:
ok this is how your eeproms look like (only the charging current ADC calibration): 
1. 50w:
```
const AnalogInputs::DefaultValues AnalogInputs::inputsP_[] PROGMEM = {
...
    {{482,  50},         {14275,  1000}},   //Ismps
...
}
```
2. 60w:
```
const AnalogInputs::DefaultValues AnalogInputs::inputsP_[] PROGMEM = {
...
    {{2713,  200},         {28498,  2000}},   //Ismps
...
}
```

1. for 50W: we measured (ADC reading) 482 for 50mA, and  14275 for 1000mA
2. for 60W: 2713 for 200mA and  28498 for 2000mA

You probably saw these values during calibration.
From this two points we can estimate (extrapolate) what the current will be when the ADC reaches it's 
maximum reading (about 65400):
1. 50W:
http://www.wolframalpha.com/input/?i=%7BFit%5B%7B%7B482,++50%7D,%7B14275,++1000%7D%7D,%7B1,x%7D,x%5D+,+x%3D65400%7D
  maximum measurable current is 4521mA
2. 60W:
http://www.wolframalpha.com/input/?i=%7BFit%5B%7B%7B2713,++200%7D,+++++++++%7B28498,++2000%7D%7D,%7B1,x%7D,x%5D+,+x%3D65400%7D
  maximum measurable current is 4576mA

In both cases the current is about 4.5A. My conclusion is that you charger is only able to charge up to this value and charger is out of specs, unfortunately this happens very often, that is wy we have the calibration error codes.

dwc56, 2018-01-07:
so with the firmware that came on the charger could it only charge with 4.5A

stawel, 2018-01-08:
yes, it looks like so,
(or  your charger was somehow damaged,
or your ammeter is damaged)

closed 2018-01-13 by stawel