stawel/cheali-charger #248: Support for Nuvoton NUC029LAN

tremez, 2018-01-08:
Just got a new animal with this "new?" chip.
Is it supported ? Looks like it is same Cortex-M0 as 0519M

stawel, 2018-01-13:
unfortunately it's not supported

Stormwing01, 2018-02-13:
When Will it be supported?

flaksirus, 2018-04-01:
@Stormwing01 I don't think it will be ever supported - it's absolutely different mcu with different architecture, and obviously there should be absolutely different schematic, so it would really hard to port this firmware to nuvoton analog.

pampyras, 2019-10-23:
Hi. Any progress on nuc029lan support?

diggit, 2019-10-24:
There is no active developer at this moment. I guess, that contributions are welcome. 

Where did you get charger with this MCU?

pampyras, 2019-10-28:
From Aliexpress. I have made it semi working by adapting code, but still need some improvents to make it usable.

shadowlab, 2021-06-04:
https://github.com/stawel/cheali-charger/issues/241#issuecomment-855119485

I've made a post in another thread regarding this CPU and charger. Hopefully, this will work OK.