stawel/cheali-charger #250: maxPC maxPD v_a0x1001_i0x6_10_3_12.py

AndyILC, 2018-01-13:
maxPC & maxPD not defined in class Settings(Structure) (v_a0x1001_i0x6_10_3_12.py)
eeprom.py incorrect parsing eeprom.bin

stawel, 2018-01-13:
you are using the eepromExtractor, nice!

first of all the v_a0x1001_i0x6_10_3_12.py files are generated automatically,
the 10_3_12 numbers refer to the eeprom version, defined in CMakeList.txt:
```
set(cheali-charger-eeprom-calibration-version 10)
set(cheali-charger-eeprom-programdata-version 3)
set(cheali-charger-eeprom-settings-version 12)
```

unfortunately when @ChihYing_Lin added the maxPC & maxPD to Settings:
https://github.com/stawel/cheali-charger/commit/fcef914e4c4522502c8c02e72a726c48b680b8ca

I forgot to increase cheali-charger-eeprom-settings-version to 13, and now you are
not able to use eepromExtractor for your eeprom. What's the solution:
1. update your cheali-charger repo to the develop branch and flash your charger with newer firmware 
    - in this branch eeprom version is 10.3.13 (with maxPC, maxPD and additionaly maxVc defined)
2. generate a newer v_a0x1001_i0x6_10_3_12.py - unfortunately this is quite hard to do

AndyILC, 2018-01-13:
1. it's probably easier to add 2 lines to the .py

--- v_a0x1001_i0x6_10_3_12.py.ori	2018-01-14 05:00:35.927468200 +1100
+++ v_a0x1001_i0x6_10_3_12.py	2018-01-14 03:02:12.771482000 +1100
@@ -135,6 +135,8 @@
     ('maxIc', ValueType),
     ('minId', ValueType),
     ('maxId', ValueType),
+    ('maxPc', ValueType),
+    ('maxPd', ValueType),
     ('inputVoltageLow', ValueType),
     ('adcNoise', uint16_t),
     ('UART', uint16_t),

and in these files too:
v_a0x1002_i0x6_10_3_12.py 
v_a0x1002_i0x8_10_3_12.py 
v_a0x4001_i0x6_10_3_12.py

2. development branch have intresting string MAX_BANANCE_CELL

3.  HardwareConfigGeneric.h have intresting definition ENABLE_B0_DISCHARGE_VOLTAGE_CORRECTION
Do you have plans to solve this problem?

stawel, 2018-01-13:
regarding 1: yes that's also an option, unfortunately I can't merge this since there are two different eeproms with the same version
regarding 2: fixed

>HardwareConfigGeneric.h have intresting definition ENABLE_B0_DISCHARGE_VOLTAGE_CORRECTION
Do you have plans to solve this problem?

unfortunately I don't plan to to this, I don't have the free time to work on cheali-charger,
but I highly encourage you to try it yourself, any forks and new projects derived from cheali-charger
are very welcome