stawel/cheali-charger #251: Feature Request: Lipo Break-in

LightningStalker, 2018-04-11:
The [recommended lipo break-in procedure](http://s3fpv.com/wiki/index.php?title=Conditioning_New_Batteries) has a many steps and I have used it but it's very tedious. This is something all chargers should have but none of them do. What do you think?

stawel, 2018-04-11:
Interesting, never heard of this "break-in" procedure, do you know any reasonable research papers on this topic or is a battery manufacturer recommending this procedure?

LightningStalker, 2018-04-12:
I'm not aware of any manufacturers recommending this exact procedure, but Thunder Power for example recommends no more than a 3-5C discharge for breaking in new packs. (http://www.thunderpowerrc.com/Support_3/Chargers_5)

specing, 2018-04-12:
http://batteryuniversity.com/learn/article/how_to_prime_batteries

l29ah, 2020-01-26:
So what's the capacity increase that is expected to be obtained that way?

TBAMax, 2022-12-25:
It seems just to enable the cycling capability for Lipo batteries. I have noticed too that some Lipo packs need 2-3 cycles after storage to return to the good working order. So yeah, cycling for LiPo that is.