stawel/cheali-charger #252: data logging software

dwc56, 2018-04-11:
i have a imax b6 and a ftdi converter connected to the temp port i wand to make a discharge curve from a battery what software can i use to do that?

dwc56, 2018-04-11:
i tried LogView and could not get it to work how do i use cheali-logviewer

stawel, 2018-04-12:
you have to save the UART output to a file and just run:
$ ./cheali-logviewer your_file
(works on linux)

> i tried LogView and could not get it to work...

First of all, in all cases, you need to be able to see the chargers output 
as a human readable text (use "screen" on linux or "PuTTY" on windows).

You probably tried "LogView studio" it's not the same as LogView,
some people where able to run cheali-charger with "LogView studio"
you will probably find some info on the chali-chargers mailing list.

dwc56, 2018-04-12:
this is what logveiw shows in the serial logger other programs like PuTTY work the charger is set to 57600 baud rate is that the correct baud rate?

07:58:29,398 [A] jº¥kUjíTÕUµµµµµªÑ+U¹ÕTQ‹-µ··µµµµµµ··«-µº¢ü
07:58:30,256 [A] ª·º­µ«ÿTÕ¥-µµµµªQ+U©ÕTQ‹-µµ··µµµµµµ·«-µM¢ü
07:58:31,041 [A] jª¥kU«ÕTU¥-µµµµºQ+U©ÕTQ‹-·µµµµµµ··µµ«-µU¢ü
07:58:31,825 [A] jª¥kUªûTU¥-µµ·µªQ›<NAK>©ÕtÑ‹-µµµµµ··µµµµ«-µªø
07:58:32,609 [A] jªSû­JíTÕU-µµµµªÑ+U¹ÕTQ‹-µ··µµµµµµ··«-µ]¢ü
07:58:33,334 [A] jª¥k½2íTU¥-µµµ·ºQ+U©ÕTQ‹-·µµµµµµµ·µµ«-·Õ¢ü
07:58:34,100 [A] ªµº­­ÕTÕ¥-µµµµªQ+U¹ÕTQ‹-µµ··µµµµµµ·»-µÅ¢ü
07:58:59,285 [A] jº¥ûŠÕ™ûTU¥-·µµµµµª”Õ‹-µµµµµ··µµµµµµ«-µUå
07:59:00,144 [A] ªµª¿QµUÿTUå-µ··µªQ›<NAK>©ÕtÑ‹-µµµµµµ··µµµ«-µŠ¢ü
07:59:00,869 [A] êµª_Qj©ûTU¥-µµµ·ÝT¥Eªµµµ··µµµµµµµ··µ«-·Š¢ü
07:59:01,729 [A] jªSû<DC4>UªûTU¥-µµµ·ºQ+U©ÕTQ‹-··µµµµµµµ··«-µª¢ü
07:59:02,512 [A] jº¥û<DC4>½¢ûTU¥-·µµµªQ+U©ÝtQ‹-µµµµµ··µµµµ«-µª‹ø
07:59:03,318 [A] ª·ª¿Ñµ¯ÿTUå-µ··µªQ›<NAK>©ÕTÑ‹-µµµµµµ··µµµ«-µ<NAK>å
07:59:04,006 [A] ªµª¿Q­ªÿTÕ¥-µµµµÕt¥E®µµµµµ·µµµµµµ··µ«-·Uå
07:59:04,792 [A] ªµª¿Ñ­ÕTU¥-·µµµªQ+U©ÝTQ‹-µµµ··µµµµµµ«-µUå
07:59:12,274 [A] jªSûTWªûTU¥-µµµ·ºQ+U©ÕTQ‹-µµµµµµµ·µµµ«-µª¢ü
07:59:13,446 [A] êµª_¥­µTU¥-µµ··ªQ+U©ÕTÑ‹-µµµµµµ··µµµ«-µªå
07:59:14,231 [A] jª¥û*™zíTUå-µ··µÕTSEªµµ·µµµµµµµ··µµµ«-µªå
07:59:15,023 [A] ªµª¿¥kUíTUå-µ·µµªQ›<NAK>©ÕtÑ‹-µµµµµµ··µµµ«-µªå
07:59:15,873 [A] ª·º¿U­ÕTU¥-µµ··ªQ+U©ÕTQ‹-µµµµµµµ··µµ«-·«å

fiveangle, 2018-04-12:
And I’ll add another: DataExplorer

stawel, 2018-04-12:
>  is set to 57600 baud rate is that the correct baud rate?

it looks like it is, see:
https://github.com/stawel/cheali-charger/blob/master/utils/LogView/ChealiCharger_6cell_1.0.ini#L35
you can also try other baud rates, just change line 35 in the ini file.

>07:58:29,398 [A] jº¥kUjíTÕUµµµµµªÑ+U¹ÕTQ‹-µ··µµµµµµ··«-µº¢ü

this is definitely not human readable....

Brassee, 2019-01-26:
Hi!
I have same problem. In Arduino IDE port monitor i get nice readable string and in LogView i get something strange like what Ghost have. Maybe something with encoding? I think LogView can't understand ASCII correctly.

berylleigh, 2020-01-08:
Logview has you sorted https://youtu.be/rwSv8El2ahg
 http://www.logview.info/forum/index.php?pages/eng/

xa0c, 2022-01-14:
Sorry for the off-topic, but I have such question:
Will it be possible to monitor battery temperature through the temperature port and at the same time monitor/log all values (current, voltage and temperature) through this temperature port?

stawel, 2022-01-14:
@xa0c 
>Will it be possible to monitor battery temperature through the temperature port and at the same time monitor/log all values (current, voltage and temperature) through this temperature port?

it is possible but you have to modify your charger,
see: https://github.com/stawel/cheali-charger/blob/master/docs/settings/settings.md#uart-settings
"separ: temperature connector and UART are separated"