stawel/cheali-charger #253: What usb serial adapter can be used?

shkup, 2018-04-15:
The documentation assumes we are all Arduino programmers...
What serial usb adapter can be used in the flashing process? 
All I've seen have RX, TX pins... How does it connect to the IMAX b6 board?
Do you have maybe some guide?

specing, 2018-04-15:
Anyone can be an Arduino programmer, it is child's play.

Any 5V ones will do. 
5V power goes (for serial adapter, don't connect it if it is already powered on its own) closest to input power connector,
then RX in middle and GND.

shkup, 2018-04-17:
But the picture in the flash tutorial states that the connection for flashing is done via MOSI, MISO, SCK etc.
How do you connect a regular serial adapter with RX, TX into it?

specing, 2018-04-17:
but you are not flashing ?! USB-serial is for saving logs to computer. You cannot flash cheali-charger with an usb-serial adapter.

Smeat, 2018-04-17:
Search for `usbasp` to find a suitable programmer.

shkup, 2018-04-18:
I want to flash cheali firmware into my b6 skyrc charger...I assume I need to connect it to the pc using serial adapter and then flash the hex file using appropriate software..Am I wrong?

specing, 2018-04-19:
yes, you are wrong.

shkup, 2018-04-20:
Can you please kindly explain?

Smeat, 2018-04-20:
Just connect MISO, MOSI, SCK, RESET, VCC and GND to the usbasp (or any other programmer), install avrdude and flash the hex using the following command:
`avrdude -patmega32 -cusbasp -Uflash:w:cheali-charger-[VERSION].hex:a`

For further information see: https://github.com/stawel/cheali-charger/blob/master/docs/flashing.md

specing, 2018-04-21:
It helps to read documentation every once in a while :)

shkup, 2018-04-22:
Just  the usbasp part was missing... Someone has recommendation for usbasp that is shipped worldwide?