stawel/cheali-charger #256: "Weak" UART on IMAX B6?

specing, 2018-05-03:
I've noticed that none of my usb-uart adapters but the MAX232 are working. The DC voltage between power and gnd is 5.04V, between signal-gnd 4.2V. Watching it on the scope I notice that the voltage difference is only around 3V when the usb-serial adapter is not connected, but drops to ~0.4V when it is connected. Are there some resistors between mcu and the rx/middle temp probe connector?

Unfortunately all schematics I've seen only document the power parts, not the temp probe. Or they include an usb-uart which should not be there.

specing, 2018-05-06:
nvm found https://groups.google.com/forum/?_escaped_fragment_=topic/cheali-charger/LKy6qITaVDg#!topic/cheali-charger/LKy6qITaVDg

Such a sad end for issue number (1<<8).

specing, 2018-05-12:
Bypassing the middle one of the group of 3 resistors does not help. Guess there is something else underneath the LCD module or excessive resistance on the return path as well. Will have to desolder the former and measure at mcu.

stawel, 2018-05-12:
This is how it looks like on one of my chargers:
[image]
schematic:
[image]
charger:
https://github.com/stawel/cheali-charger/blob/develop/docs/imaxB6/imaxB6-clone-front.jpg

unfortunately each charger has its own circuit.
Hope this helps
Paweł

specing, 2018-05-26:
Thank you, it has helped massively. Looks like I've been bypassing the wrong resistor.

Turns out that bypassing the 39k one did work (it took a while to position a 4mm long piece of stranded wire and solder it over the resistor as I had lost track of the spare 0403-or-some-other-matching-tiny-one). I'll run one operation and see if the checksums match.

specing, 2018-05-27:
All checksums match in recording of 1h30 of charging @ 57600baud.

specing, 2018-08-29:
Are there some schematics floating around for the connections between the atmega32 and LCD? I would like to modify mine to receive data as well. I want to see if obtaining a bidirectinal uart connection is feasible or if I should look into some other protocol.

I could not find the cheali-charger group that is available via googlegroups in usenet via eternal-september.org. Is this a usenet or some other form of group?

neolead, 2020-10-15:
thank you very much. did this with imax b6ac+ ,,, original...