stawel/cheali-charger #258: How to build the source for atmega32 - windows

EKURRU, 2018-06-03:
Hi,stawel:
   I readed the  building source guide as below.
could you pls point it out more clear this sentence“s:\cheali-charger> cmake . -DCMAKE_TOOLCHAIN_FILE=avr-toolchain.cmake -G "Unix Makefiles"”，waiting your reply！thx！

Atmel Studio

install Atmel Studio and cmake
start a "Atmel Studio Command Prompt" and go to your cheali-charger directory.
run:
  s:\cheali-charger> cmake . -DCMAKE_TOOLCHAIN_FILE=avr-toolchain.cmake -G "Unix Makefiles"
  s:\cheali-charger> make
hex file should be in s:\cheali-charger\src\atmega32\targets\[your charger]\cheali-charger*.hex

stawel, 2018-06-07:
hm... I'm not a windows user but I'll try

1. install Atmel Studio
2. install cmake
3. "start a "Atmel Studio Command Prompt"  - 
   "Atmel Studio Command Prompt" should be provided with Atmel Studio, it's a cmd.exe with some 
   paths set
4. "... and go to your cheali-charger directory." - just type in "Atmel Studio Command Prompt":
    `cd c:/path/to/your/cheali-charger`
    in my example the path was _s:/cheali-charger_
5. sentence: “s:\cheali-charger> cmake . -DCMAKE_TOOLCHAIN_FILE=avr-toolchain.cmake -G "Unix Makefiles" " - just type in "Atmel Studio Command Prompt":
   `cmake . -DCMAKE_TOOLCHAIN_FILE=avr-toolchain.cmake -G "Unix Makefiles"`
  if you get an error, something like _cmake command not found_ you have to add to %PATH% the cmake.exe path (see 2), just type:
   `SET PATH=%PATH%;"C:\Program Files\path\to\cmake\check\where\cmake.exe\was\installed\" `
   and repeat the cmake command.
6. type:
   `make`
7. your new hex file should be in _c:/path/to/your/cheali-charger/src\atmega32\targets\\[your charger]\cheali-charger*.hex_

specing, 2018-06-07:
I thought it would start by 1) Install GNU/Linux :D