stawel/cheali-charger #260: Take too long time for charging and balancing

michael-blue2, 2018-08-01:
Most of the times a normal charge + balance on my stock fw it would take around 40-50min max. With the cheali v2.00 it always at 1h(charging at 1c, 4s 1300mah, intern 84, 50, 33, 23 ) But sometimes that charge time might take even a lot of hours. Just came to my charger and it was writing 335min to balance, 3 cells were 4.00 +- and the other one which i replaced was 3.85+-, and this charging took 5h to balance, how is that even possible?

stawel, 2018-08-01:
To investigate this issue, I need a LogView log of your charging process.
If you do have a UART->USB dongle, please do a "charge + balance" with 
the dongle connected, UART enabled (settings -> UART: ON).
I only need a raw UART dump so if you don't have LogView configured, any
serial terminal dump will do.

But in general it is possible that the charge + balance process takes a lot of time,
usually when the battery is damaged (but probably not in you case).

Other options you may try:
1. change [battery] -> "edit battery" -> "bal. err:" to something bigger (~20mV)
 or
2. don't use "charge + balance", use "charge" and after that use "balance"

michael-blue2, 2018-08-02:
Thanks a lot for you fast reply, I own a Tenergy Tb6b( 2 of them both running latest version, with the imax b6 clone hex file) I did flash them with a usbasb and i dont thing so that theres a uart port. I have just read some stuff about connecting it to the 3pin on the side for the temperature but i dont have any other informations on the pinout

michael-blue2, 2018-08-02:
https://www.dropbox.com/s/zui4mz3iqlz5vyw/Photo%2002-08-2018%2C%209%2000%2016%20AM.jpg?dl=0
This is a picture of my temperature pins

stawel, 2018-08-03:
Yes, your charger probably has a UART output on temperature pins,
this should help:
https://github.com/stawel/cheali-charger/blob/develop/docs/imaxB6/imaxB6-clone-temp-TX-schematic.png
you only need to connect TX (to RX on dongle) and GND.

In software set baudrate to the same as in settings, communication is 8N1 (8bits, No  parity, 1 stop bit)
Try to dump the serial communication  to a terminal,
on linux you can try: minicom -D /dev/ttyUSB0
on window: you can probably use PuTTY.
You should see a nice human readable text.