stawel/cheali-charger #261: strange mode of failure / imax b6 80W atmega32

goahead0, 2018-08-10:
Hello,

For the last 10 days or so i have been extensively and happily using my imax b6 80W charger with cheali-charger-Dual-Power-B6AC-80W-RC_2.00-e10.3.12-20160613_atmega32.hex. During this time i have been constantly using the charger,  testing, charging, logging all my collection of batteries without an issue.

Yesterday the charger stopped working in a very strange way. While balance charging the last battery pack, i noticed that the buzzer sounds were much less loud as usual and the sound was distorted in a strange way, almost as if the buzzer was driven with too low a voltage. The charger was still charging, the display still working.

To check what's wrong, i disconnected the battery pack from the balancing and charging leads and unplugged the power supply. I checked the power supply, (a PC power supply providing 12V), which was ok. The battery pack being charged seemed to have trouble (a 2S 100mAh Lipo), the middle balancer wire had become loose and was causing the display of the cell resistance of one cell to fluctuate between 27mOhm and over 10Ohm.

After plugging in the charger again, i only see a row of squares on the display and there is no reaction when pushing the buttons, no sound at all from the buzzer.

I thought that maybe the controllers flash had become corrupted for a strange reason, so i connected the charger to the USBASP dongle, reprogrammed the atmega32 with the hex file i have been using for the last 10 days (from the zip file of the master branch) with avrdude, after checking that the sha1sum of the hex is correct. The controller can be programmed and verified without problem.

I checked that the fuses are correct. Unfortunately after powering up the charger i am still getting the row of squares on the brightly lit display and no other sign of life from the charger. I tried a different hex files, with the same result, i flashed the original flash, eeprom and fuses, the same result. I checked the push buttons, they work and none is stuck.

To sum it up: 
- the charger was working great with cheali-charger
- in the middle of a charging operation started acting strangely
- after unplugging and replugging power, just a bright display with squares, otherwise dead
- can be reprogrammed (correct hex, correct fuses), issue remains

If anyone has ideas what could be wrong or how i could investigate the problem, please tell. I'm grateful for any hints.  If i can be certain that the controller is broken, I'll order some chips and replace the atmega32.

Thanks.

stawel, 2018-08-12:
> Meanwhile i could confirm that there is a clean looking 16MHz sine wave with an amplitude of 0.68V at the crystal pins.

that's a good start, first you should determine if it's a CPU malfunction:
1. check if VCC is 5V (pins 5,17,27,38)
2. check if GND is 0V (pins 6, 18,28,39)
3. check if ~RESET pin is 5V (pin 4)
4. check XTAL1, XTAL2  signal (pins 7,8) - DONE

make sure you are measuring the voltage directly  on pins (or as close as possible)
some of the cheap chargers have very week PCB traces (vias), they may be broken.

if the CPU is working correctly and the buzzer is ok you should hear beeps when you press a button.

goahead0, 2018-08-12:
Thank you for the ideas. So far i had checked the -RST at the isp connector and it was at +4.9V, the same level as VCC. 

I'll remove the LCD to get access and check the signals at the processor directly. And also inspect the soldering and see if i can locate  PCB weaknesses. I too noticed that the tracks look quite flimsy.  

I'll come  back after completing your suggestions.

goahead0, 2018-08-13:
In the meantime i removed the display and did following measurements:

VCC pins 5,17,27,38 are 5V
GND pins 6,18,28,39 are 0V
RST pin 4 is 5V

crystal pins 7,8 are DC4.7V, AC2.5V with multimeter (AC range is very coarse)
with oscilloscope there is a 16MHz sine wave on both pins with an amplitude of about 1.8V

The connections were checked between processor and LCD, processor to buzzer transistor base resistor (R56 on the RC power BC6 schematic), they are ok. I didn't observe any broken vias or PCB defects during a short check.

I also read your thread about the fixing attempt of a 200W charger with broken vias and the very interesting link about failure modes of vias.

goahead0, 2018-08-13:
Hello again

Something odd that i noticed was that i can flash the atmega only with 3.3V. My programmer allows to select a supply and signal voltage of 3.3V or 5V. 

When selecting 5V i was unable to access the atmega and the first row of the display was filled with squares. The same also happened when letting the charger supply itself (not connecting VCC from the programmer with the charger, powering the charger via its DC connector).  I can't see why it shouldn't be possible to flash the controller with 5V, after all that's its usual VCC.

When selecting 3.3V, the display was empty and i was able to flash.

stawel, 2018-08-13:
could you check your CPU label?
from atmega32 datasheet: 
```
*Operating Voltages
– 2.7 - 5.5V for ATmega32L
– 4.5 - 5.5V for ATmega32
*Speed Grades
– 0 - 8 MHz for ATmega32L
– 0 - 16 MHz for ATmega32
```
some manufacturers put ATmega32L into the chargers with a 16 MHz oscillator,
it usually work, but probably not always.

goahead0, 2018-08-19:
Thanks, i have ordered 2x Atmega32 au a couple days ago. I'll let you know how it goes.

goahead0, 2018-08-23:
The charger is working again! Before replacing the processor, i thought that i could touch all its pins with the soldering iron to make sure that they had a good contact with the board. That's all that was needed to get the buzzer to react to pressing buttons again. The actual work was connecting the display with the board with 16 short pieces of wire, so i still had access to the processor in case that it would need to be replaced. Today i also received the second charger that i had bought to replace the first one. Good timing.

It looks like at least one of the processor pins was badly soldered and eventually became disconnected from the board.

Thanks a lot for your help and for all your time you put into this project.

goahead0, 2018-09-01:
I noticed that especially with prolonged discharge at much over 5W dissipation even with external fan cooling the processor clock over time seems to  shift enough to first cause transmission speed errors on the telemetry uart.  Then the buzzer sound frequency gets lower and lower. After some time the controller freezes, the LCD is stuck with the last displayed values, no reaction to button presses. I can try using a crystal with lower temp coefficient but this would only fight symptoms instead of being a real solution i suppose.

Since this issue developed and got worse over time i looked at the spot heating up the enclosure, which was near the battery output connectors. The mosfet which is  located there and which switches the current from battery - to gnd discharge resistor, had some slight discoloration on the cooling fin, probably caused by heat. I have replaced that IRF740 with an IRLZ34. 

The IRLZ34 has a much lower rdson, 28mOhm instead of the IRF740's 550mOhm. The charger seems to work more reliably now but in retrospect i think that the higher rdson could be better in this application because the MOSFET is cooled by the case, whereas the discharge resistor is underdimensioned and only has air cooling inside the case. Because of the lower rdson the discharge resistor will have to dissipate a higher proportion of the discharge power.

I think a more proper solution would be to have the discharge fet and resistor outside the case on a proper heatsink to allow higher discharge currents without thermal issues. Until than it's probably wise to limit the discharge power to levels around 5W.

diggit, 2018-09-01:
Hi @goahead0,
Different discharge transistor won't help you, it may even cause problems.
Discharge circuit keeps transistor in resistive area, **it is never fully turned on**. Thus Rds_on is meaningless.
Resistor in discharge circuit is only for current sensing and should dissipate minimum power.

So if you want improve cooling during discharging, place that transistor on heasink.

goahead0, 2018-09-02:
In the meantime i too realised that the discharge transistor is not used in switching mode and that the "discharge" resistor is only there for measuring the current.

I'll limit the discharge dissipation power for now and maybe later move the transistor out of the case.

The other symptom with swinging clock speed was indeed caused by a bad crystal. After replacing it that issue is gone.

Thanks a lot.

closed 2018-09-05 by goahead0