stawel/cheali-charger #265: Kod programu 32kB nie mieści się w pamięci 60kB ?

hotair1983, 2018-10-12:
"After all of that  I was able to compile the code but  the code size was about 35kB. ..."

"nevertheless It looks like it should be possible to run cheali-charger on MA84G564,
but I hoped to go down with the code size below 32kB so it could be run also
on other mcs-51 CPUs."

_Originally posted by @stawel in https://github.com/stawel/cheali-charger/issues/106#issuecomment-222520952_

Witaj Paweł:

Nie znam się na programowaniu dla tego wolę zapytać.
Skoro po skompilowaniu kod programu zajmuje około 35kB to dlaczego jeszcze bardziej chcesz go zmniejszyć skoro pamięć w układzie MA84G564 dla programu aplikacji przeznaczona jest do 60kB ?
MA84G564 Flash Memory:
https://obrazki.elektroda.pl/3407102900_1539345440_thumb.jpg

Można prosić o ten skompilowany kod 35kB chciał bym sprawdzić czy dało by rade go wgrać poprzez USB DFU (Device Firmware Upgrade)

MA84G564 Evaluation Stick User Manual :
https://mega.nz/#!mpFi0A5a!Mi_TnjzuEdAMRkpP-7oiECJDcrfXXIOwk6u3bMCPfEM

hotair1983, 2018-10-12:
Inny projekt czyli specjalnie pod architekturę mcs-51 pisany nowy kod program a przynajmniej z tymi MCU spełniającymi wymogi te minimalne na działanie  cheali-charger'a czy to raczej już całkiem dostosowany pod dany MCU. Ogólnie zdaję sobie sprawę że to jest na pewno czasochłonne i wymaga nie przeciętnej wiedzy ale jak by była chodź mała szansa na sukces takiego projektu to może i by było warto. Aktualnie bo kto wie czy nie więcej na MCU MA84G564 udało mi się odszukać trzy ładowarki:
1)TURNIGY ACCUCEL 6 80W
2)SkyRC iMax B6 Mini
3)VOLTCRAFT  UCT 100-6

Z opisów na jakie napotykałem owe ładowarki występują też z innymi MCU np Atmege32... te z MA84G564 itp nazywa się klonami nie wiem dlaczego może to podróbki albo cenowo producenta taniej wynosi stosowanie tych układów na to pytanie nie znam odpowiedzi.

Co do VOLTCRAFT  UCT 100-6 to akurat wiem że występuje w 2 wersjach ja mam tą niby wersję klon.
Ostatnio mając problem z komunikacją ładowarki z programem na PC dedykowanym ChargerMonitor wgrałem poprzez DFU wsad od SkyRC iMax B6 Mini i po części strzeliłem sobie tym w kolano bo jedynie logo startowe się pojawia...w dalszej części natrafiłem na oprogramowanie do TURNIGY ACCUCEL 6 80W i o dziwo to ruszyło podświetlanie zapala się na tylko logo startowym ale menu działa lecz i tak nie do końca bo wiadomo program nie był pisany pod VOLTCRAFTA. Pisałem do dystrybutora i producenta o wsad ale to mało prawdopodobne abym go otrzymał. Tak więc jak by co to mogę nawet podesłać tą ładowarkę jak by tylko miało to pomóc i przyśpieszyć w tego typu projekcie bo tak czy siak robi mi teraz za przycisk do papieru... Ciekawi mnie tylko jedno czy jak bym miał taką samą ładowarkę czy da się z takiego układu zgrać program jako wsad i wgrać do mojej ładowarki poprzez DFU o ile nie ma jakiejś blokady itp.
https://www.mikrocontroller.net/topic/139664

stawel, 2018-10-13:
>Inny projekt czyli specjalnie pod architekturę mcs-51 pisany nowy kod program a przynajmniej z tymi MCU spełniającymi wymogi te minimalne na działanie cheali-charger'a czy to raczej już całkiem dostosowany pod dany MCU. 

raczej cały nowy projekt specjalnie pisany pod tą architekturę, niestety architektura jest dość specyficzna, choć pewnie dałoby radę przerobić cheali-chargera tak by na niej działał (oczywiście procesory z ~32kB ROM nie wchodzą w grę).

Niestety, ja nie mam wystarczająco wolnego czasu by się tą architekturą zajmować (mcs-51 jest bardzo stare, ma już ponad 35lat, inaczej piszę się pod nią programy niż we współczesnych rozwiązaniach). Nie mam nawet czasu by poprawiać błędy zgłoszone do aktualnej wersji cheali-chargera
(za dużo pracy zawodowej mam na głowie), więc ktoś inny by musiał się tym zająć.

hotair1983, 2018-10-13:
Dziękuję za informacje to co stworzyłeś to i tak sporo. Na pewno nie jeden jeszcze ciekawy projekt w twojej głowie się zrodzi z mojej strony mogę życzyć tylko powodzenia i dużo zdrowia i wolnego czasu dla siebie i bliskich.