stawel/cheali-charger #266: Imax B6 clone unsupported

JLIspace, 2018-10-23:
Hey,

I flashed my imax b6 clone with the correct version of cheali-charger. It's powered by a ATmega32A and it did flash without any errors. I also verified the flash/eeprom afterwards and it looked fine.
The problem being that the display only shows blocks and the charger beeps once after startup.
It's probably not supported I guess? Or is there anything left I can try?

Also I failed backing up the previous software version so I'm stuck with a bricked charger for now. I'd be very grateful if anyone could help me with this. Maybe someone could send me his backup? :)

Here are some pictures from the PCB:
https://photos.app.goo.gl/cCHJRvYEfPiFyrUv5

Thanks in advance 👍

stawel, 2018-10-24:
Hi JLIspace,
have you tried:
https://github.com/stawel/cheali-charger#troubleshooting

`2. Sha1 sum is correct and the charger still doesn't work (display shows squares):`

JLIspace, 2018-10-24:
Hey,

well.. I only use GitZip for downloading files on github so that shouldn't be a problem. Anyway, I reflashed it while making sure the sha1 checksum was correct. Unfortunately the result was the same. It's only displaying blocks an it is not beeping when pressing buttons.

I used these files:
cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.hex
cheali-charger-imaxB6-clone_2.00-e10.3.12-20160613_atmega32.bin

stawel, 2018-10-24:
have you tired to reset the fuses?
(sorry I should have copied the full text:)

2. Sha1 sum is correct and the charger still doesn't work (display shows squares):
- reset atmega32 fuses using avrdude:
   - windows: avrdude.exe -patmega32 -cusbasp -Uhfuse:w:0xc5:m -Ulfuse:w:0x3f:m
   - linux: avrdude -patmega32 -cusbasp -Uhfuse:w:0xc5:m -Ulfuse:w:0x3f:m

JLIspace, 2018-10-24:
Why would I want to disable SPI. Isn't usbasp literally relying on it?
Also I never managed to make it answer again when setting it to use external osc. so I set it to internal clock. Could this be the problem?

closed 2018-10-24 by JLIspace