stawel/cheali-charger #267: Compatibility - different revisions of Turnigy Accucell 6 50W 6A

aShure, 2018-12-08:
This Accucell 6 50W 6A PCB here does not seem to match the documented one. I can figure out the ICSP port, but is the rest of the HW compatible? There are less balancing resistors for example.

[image]
[image]
[image]

stawel, 2018-12-09:
It's really hard to tell, fist of all you should check the cpu, it should be atmega32 or nuvoton-M0517,
the rest of the HW can be adjusted in the software if it changed.

> There are less balancing resistors for example.

not really, you have 2x43 ohms in parallel, that's 21.5 ohm per channel, in my imax B6 
I have 6x120 ohm = 20 ohm per channel (it's very close), although the Watts are probably different,
I have 1206 resistors, what gives us 6*0.25W ~= 1.5W per channel, You probably have 0.5W per channel (or 1W depends on the resistors diameters), and you need at most 4.2V^2/21.5 ~= 0.82W so your resistors are a little bit to small, but we do some duty cycling when balancing so it may be enough, 
if not, the resistors should get to hot, In this case we may need change the duty cycle in software.

v0l0d, 2020-10-06:
Can anyone please help identify this charger and it's pins?
I used sequence: VCC, GND, Reset, MOSI, MISO, SCK for the command 'avrdude -patmega32 -cusbasp' but I get error 'avrdude: initialization failed, rc=-1'

Thank you!

diggit, 2020-10-07:
Hi @v0l0d, as Stawel said, it's difficult without knowing what MCU your charge has. Please provide photo of it. MCU is located below LCD. You can carefully bend LCD up to see MCU marking. Only atmega32 a nuvoton-M0517 targets are supported.

v0l0d, 2020-10-07:
@diggit,
Thanks for the quick reply
Unfortunately I cannot bend the panel. It looks to be soldered and screwed on :(
Does avrdude support Nuvoton-m0517 ?
I’ll try it
Options “  -patmega32 -cusbasp ” don’t work; once I run avrdude just with these option my charger changes menu (like I would press a button). Then avrdude shows the error :(

diggit, 2020-10-07:
Once you unscrew bolts holding LCD it should be possible to bend header which connectes it to main board (with some force) a little bit, to see marking on MCU. 

Avrdude is tool for AVR MCUs (family, that Atmega32 belongs to). Nuvoton-m0517 is completely different MCU. Programming protocol is not compatible at all.

You really need to know, what MCU sits on your board. There is a chance, that mcu on your board has no support in cheali.

gigawatts, 2023-10-10:
I've got the same board revision, I lifted the LCD and the micro-controller chip under has no markings on it at all, so probably a clone of something. The 6 pin header across the bottom connect to 5v, GND, and each of the 4 push buttons. shorting each of those 4 pins to ground causes a button push event. This is not the ICSP header.

There are 3 through-hole test points under the LCD, labeled TP1, TP2, and TP3 on the backside of the PCB. I wonder if these might be a SWD programming interface?

byRoG-174, 2024-05-26:
MCU Atmega406
Key: VCC-1. GND-2, PA4-3, PA5-4, PA6-5, PA3-6.