stawel/cheali-charger #270: Support ATMEG644 and ATMEG1284

jonathan2018, 2019-03-03:
hi Developer 
we had a project so good for charge on B6 by Atmega32. I hope you will support new feature ,maybe atmega32 will limit memory  ,So  i think we should move to higher chip , EX at644 and 1284,   both same pin with at32 and easy to replace .

How do you think about it ?

specing, 2019-03-06:
... What?

Maybe it will limit memory,  maybe it is Maybelline. I don't see why this project should support anything that cannot be easily purchased in a store.

diggit, 2019-03-06:
Exactly as @specing says .
Primary goal is to support stock HW.

jonathan2018, 2019-03-07:
No , that is a option for who want modify .You still can use stock Hardware but other  want update and they can buy chip to replace

specing, 2019-03-07:
If you want to use ATMEGA644 or 1284 then add the support yourself. Tbh I'm more concerned about lack of uart rx and upper battery voltage limit than code/feature space. I wonder if I could hijack any pins for the former , not to mention that someone (you?) should add character entry support to battery names. Raising the upper voltage limit would require making another board -> time for cortex-m.

jonathan2018, 2019-03-10:
> If you want to use ATMEGA644 or 1284 then add the support yourself. Tbh I'm more concerned about lack of uart rx and upper battery voltage limit than code/feature space. I wonder if I could hijack [...]

yes, i compiled successful program for AT644 with little modifi source code. now solder ic and enjoy  :))