stawel/cheali-charger #271: Settings for HV Lipos?

DerDominik, 2019-03-10:
Hi there

Until now I've used my Charger with Cheali-Charger-Firmware only with normal Lipos.
But now I've some HV Lipos, so is there something special to configure for these Lipos?

diggit, 2019-03-10:
I am not sure what is you "HV Lipo" max voltage, but there is pre-configured battery type called L435 which is just Lipo with maximal voltage (Vc) set to 4.35V. It has same effect like changing cell target voltage (Vc) to 4.35V. To change those limits manually, you have to switch menu into "advanced"

DerDominik, 2019-03-12:
I've looked for technical data for the Lipo.
It seems, that these HV Lipos have a voltage of 3.8V, so I think normal Lipo should also be ok?!

diggit, 2019-03-15:
3.8V lipos usually have 4.35 V top voltage. L435 profile should fit best. You won't fully charge them with with regular/unmodified LiPo profile, but you can still use it.