stawel/cheali-charger #272: Can't start charge for 0 volt LiIo

hryamzik, 2019-03-11:
Got some old LiIo batteries that were fully discharged and I can't start charging them as voltage is at zero. Is there any way to recover them? Original chargers are dead too so iMax seem to be the only option for now.

specing, 2019-03-11:
It is not safe to recharge a li-ion battery that was below 2V, let alone if it was left there for long periods of time.

hryamzik, 2019-03-11:
It's still possible and not having that option even in advanced mode is a bit confusing.

fiveangle, 2019-03-12:
Adding that option would imply using it is a great safe practice, which it isn't. 

Those that understand what had happened chemically when LiPo voltage drops to zero also must know how to manually address recovery since attempting to automate would be a disservice.

closed 2019-03-12 by hryamzik

hryamzik, 2019-03-12:
In this case that was a liio, not lipo battery. I'm not really aware of chemical details and risks (except overheating and catching fire) but one of these 2 batteries shows 1118mAh in a capacity check now, original capacity was 1500 mAh. Doesn't look that bad to me.

fiveangle, 2019-03-14:
Wet electrolytic Lithium-Ion formulas are in fact far more volatile than the gel based Lithium-Polymer construction so the warnings goe doubly in your case.  That said, all lithium based cell formulas carry these risks to varying degrees and what some manufacturers call "LiPo" are often in fact "LiIon" packaged in more flexible soft cell packaging, making it even more difficult to gauge if recovery is "safe enough".  Bottom line is there is no "safe" way to provide the feature you are suggesting in any automated way.  It could be possible to build an expert system that walks a user through the process, with user-driven visual confirmation at every step, in order to attempt recovery for depleated lithium cells, but the value vs. the amount of flash space necessary (which doesn't exist as flash is all but exhausted) would be of such a limited value as to not be worthwhile.

But a cell recovery _wizard_ / _expert system_ would make a nice alternative fork of the firmware specifically for this purpose, so post here if you decide to pursue that effort.

I'll try to ignore the proposal that anecdotal evidence is anything more than anecdotal, but I'm glad you were able to recover your depleated cell 👍 

-=dave

hryamzik, 2019-03-15:
Makes sense, thanks! Looks like current firmware utilizes ≈95 hardware capabilities of an ancient AVR and we need more modern hardware to put our coding efforts to.

specing, 2019-03-15:
Transplanting the chip for a pin-compatible AVR with much more flash space would be easy compared to the development work for such an "expert system".