stawel/cheali-charger #274: Turnigy TQ4 4x6 - Nuvoton M0517

demianbadin, 2019-06-01:
Hello. Hope there is some one can help with this issue.
I've flashed Cheali on a TQ4 and inmediatly it get briked. I've followed the steps from this thread:
https://github.com/stawel/cheali-charger/issues/207
(Talked with Ralim who help me in some steps)

And now I can get a partially working firmware.
Tested:
- all pinout with a tester and patience - ok
- display ok
- buttons ok
- analog inputs: ok with helper firmware / issues with charger firmware
- pwm ok (must modify outputPWM file)
- discharge ok
- charge ok

If I connect a 1 cell batt in balancer port, I get 1.7V on BAL 2, even if I make a short on BAL 2 port.
if I connect a 2 cell batt, I get 1,7V on BAL 3. Even if I make a short on BAL 3 port.
The same for 4 cells.

So, I think the problem resides in the analog read method because the HELPER firmware works fine and the charger firmware doesn't.

I saw that HELPER firmware makes real time reads from ADC, but charger firmware makes Burst reads and average from analog inputs.

Any Idea?

stawel, 2019-06-01:
Hi @demianbadin,
First, congratulations on your progress! 

> Any Ideas?

if you look at the original BC6 charger schematics (I assume your charger is very similar):
http://www.rcgroups.com/forums/showatt.php?s=df7049bcbafdb5d7d06765c264e5c4bb&attachmentid=3693125&d=1293732709

The first 2 balance port inputs are connected directly (without an opamp) to the multiplexer (4051) through R75-R76-R78-C23, R60-R61+R62+R63-R79-C25, R71-R72+R73+R74-R80-C22, and to the ADC pin trough R103-C33.
The R60,R72+R73+R74 resistances are relatively high compared to C33,
since we are switching between the inputs  quit fast the C33 has not enough time to  discharge trough  R72+R73+R74 for balance port  B2.
You also seems to have a similar issue on BAL3, maybe your charger is missing an opamp on BAL3?

Cheali-charger deals with this problem by discharging the C33 capacitor manually, see:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/AnalogInputsADC.cpp#L39
and:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/AnalogInputsADC.cpp#L134-L158

what we do:
1. interlace measurements between the MUX (multiplexer) input and "other" inputs to give C33 more time to charge/discharge
2. discharge C33 whenever we start an "other" input measurement, by:
    - switching MUX to BAL6 (so we don't discharge capacitors on channels without an op-amp)
    - start discharging C33 by setting ADC input to digital output with 0 set on this pin
    - stop discharging after ADC_CAPACITOR_DISCHARGE_DELAY_US  by switching pin back to ADC input

I assume your charger has higher resistor and/or capacitor values mentioned above.

so:
1. make sure code (2.) is working as it should, 
2. try to increase ADC_CAPACITOR_DISCHARGE_DELAY_US but as low as possible, it decreases C33 charging time.
 
Hope this helps,
Best regards

demianbadin, 2019-06-02:
Hello @stawel ! Thank you for the response.

I've checked the balancer port circuit and it differs from imaxB6 standard.
There are 6 OpAmp. (4 in the LM2902 + 2 in 2904)
All 6 cell inputs comes from an OpAmp circuit. 
Attached is a (very ugly) drawing.
There are no capacitors... The OpAmp outputs goes directly to the mux.
Also I've tested to duplicate the discharge time with no result. Maybe because there is no C to discharge?

[image]

Best regards,
Demian

demianbadin, 2019-06-02:
Just for test, I've tried to disable the discharge function by changing this lines in AnalogInputsADC.cpp

void setMuxAddressAndDischarge(int8_t address)
{
    if(address < 0)
        return;
        _setMuxAddress(address);
}

With same results. Eith 3 cell battery I get 1,569V at CELL 4 measure. Maybe is a real voltage at the analog input and can be just ignored. I must change the minimum valid cell voltage to the charger not recognise this cell as connected. I'll try 1,8V as minimum cell Voltage.

stawel, 2019-06-02:
so your chargers balance port is similar to the turnigy 2x200:
https://drive.google.com/file/d/0B1RXXTatsA1cb1R5NHM3MEtsakE/edit

This is actually a good thing (measurements should be more consistent),
and You don't need the "discharge" part of my code (it should look like your code above),

Now I'm a little confused, your charger shouldn't have these issues.
I'm also wandering why you have correct voltages when using 
AnalogInputsAnalyzer-50W, this helper uses exactly the same AnalogInputsADC.cpp code.

We probably need some more investigations:
1. make sure ENABLE_SIMPLIFIED_VB0_VB2_CIRCUIT is not defined
    - add `#undef ENABLE_SIMPLIFIED_VB0_VB2_CIRCUIT` to HardwareConfig.h
    -  to be extra safe temporary add to cheali-charger/src/core/AnalogInputs.cpp:
```
#ifdef ENABLE_SIMPLIFIED_VB0_VB2_CIRCUIT
#error "SIMPLIFIED_VB0_VB2 is still enabled"
#endif
```
2. check if AnalogInputsAnalyzer-50W is really working correctly (from my point of view, helper and charger firmware should work the same)
3. could you make voltage measurements on (symbols according to your drawing): 
    - balance port inputs BAL1,BAL2,BAL3,BAL4,BAL5,BAL6 (relative to GND)
    - op-amp outputs B1,B2,B3,B4,B5,B6 (relative to GND)
    - when no battery is connected, 1 cell battery is connected, 2 cell, 3.. , 4.., 5 ..., 6 cell battery is connected (only the balance port should be connected)
    - do this for AnalogInputsAnalyzer-50W and for charge firmware if they differ
    - write down the voltages displayed by AnalogInputsAnalyzer-50W and  charger firmware.

I know, It's a lot of measurements, it shows how confused I am ;)

> Maybe is a real voltage at the analog input and can be just ignored.

Your measurements should give us a clue, but I doubt those are real voltages.

demianbadin, 2019-06-04:
Ok. This will take me some days.
I'll prepare a 6S Lipo. (I only own 4S and 3S)
Also I'll prepare a set of wires from charger to a universal pcb to easily change test points.
Thank you for take the time to respond.

demianbadin, 2019-06-05:
@stawel , CHARGER and HELPER showed the same values...

[image]

I only write down the values with Charger firmware.

stawel, 2019-06-05:
Thanks for the measurements!

There are 3 things that worry me:
1. negative voltage on BAL 0
2. the 0.1V to 0.66V on op-amp outputs, this should be 0V  since the op-amps work as differential amplifiers (and the inputs difference is negative)
3. the ~0.68V on LCD when no cells are connected (should be 0V)
 
for 1: 
I assume You did the measurements with main battery wires disconnected?
could you connect only the negative main battery wire (black) and check if the "BAL 0" voltage goes to 0V for 6 cell batteries?

for 3:
could you measure the voltage on MUX output (4051 pin 3) and also voltage on CPUs ADC pin (not sure which pin this is, but should be connected to MUX output through a resistor.
Do this measurement without any battery connected, relative to GND and when the LCD is shows 0.675V.

for 2:
to be honest I don't have a clue, maybe  supply voltage of the op-amps is wrong?
could you measure voltages on pin 4 and pin 11 relative to GND? if you have access to an oscilloscope,  check if voltage is flat. 
If you have a different power supply  check if changes anything.

could you do a picture of your main board? (both sides if possible) 
also could you show me changes you've made to the cheali-charger code? (preferable in your repo)

demianbadin, 2019-06-05:
for 1:
I assume You did the measurements with main battery wires disconnected? 
R: yes

could you connect only the negative main battery wire (black) and check if the "BAL 0" voltage goes to 0V for 6 cell batteries?
R: yes, "BAL 0" goes to 0V when Batt- is connected

for 3:
could you measure the voltage on MUX output (4051 pin 3) and also voltage on CPUs ADC pin (not sure which pin this is, but should be connected to MUX output through a resistor.
R: the OpAmp outputs are connected through resistors to MUX inputs.

Do this two measurements without any battery connected, relative to GND and when the LCD shows 0.675V.
R: I've used Osciloscope to measure MUX output (picture 1), and then measured MUX inputs (picture 2)
It's very strange what I get... I've noticed that this pulses are bigger on MUX side than in OpAmp side (separated by the resistors)

Pic 1
[image]

Pic 2
[image]

for 2:
to be honest I don't have a clue, maybe supply voltage of the op-amps is wrong?
could you measure voltages on pin 4 and pin 11 relative to GND? if you have access to an oscilloscope, check if voltages are flat.
R: yes, pin 4 of LM2902 is Vin and is a clean power supply. Pin 11 is GND, flat 0V.

could you do a picture of your main board? (both sides if possible)
[image]
[image]

also could you show me changes you've made to the cheali-charger code? (preferable in your repo)
R: ok... I'll upload to my repo soon.

demianbadin, 2019-06-06:
I've just uploaded my files to repo.
[Compare](https://github.com/stawel/cheali-charger/compare/master...demianbadin:master)

And... can confirm that charging is not working. But is another story.
First I want to solve analog read issue.

stawel, 2019-06-06:
Interesting, I had a teorii that you are doing something additional  on ADC pin (m0517 pin 1) besides measuring  but looking from  your code I was probably wrong (the MUX (Multiplexer) is bidirectional, so everything on Mux output (pin 3) will affect its inputs)

so, I'm not sure what to do next,  here are possible scenarios:
1. we do something on ADC pin  (M0517 pin 1) in code, and I didn't cach it
2. MUX 4051 is broken
3. M0517 is broken 

you could measure the voltages on red point's (see picture) relative to ground with your oscilloscope,
with no batteries connected (I think fixing the  ~0.65V with no batteries should be our priority)
if voltage on MUX site is bigger then MUX is broken, otherwise it's something on the CPU site.
but I'm not sure if the voltage difference will be significant enough, so you could also temporary remove the resistor.
[image]

If you have any other ideas, please let me know.

demianbadin, 2019-06-06:
Hi,
ok, I'll remove resistor and measure CPU and MUX points.

Here is a picture of a complete cycle with no battery on cheali:
[image]

And, as this charger is 4 in 1, and the rest of the chargers works "fine" as original, I've measured MUX output. The turnigy firmware samples BALANCE port every 1 second. And only during operation. In idle mode the Turnigy firm does not sample balance port values.
3 cell battery:
[image]
4 cell battery:
[image]

I'm not able to test the turnigy firmware with no battery, because it doesn't start to monitor balance port.

If you look at 3 and 4 cell, there is also a 680mV after last cell.

Also, I can flash another charger. The risk is mine... of course I trust this will work... =)

stawel, 2019-06-06:
> Also, I can flash another charger. The risk is mine... of course I trust this will work... =)

this can wait ;)

Ok, I've check what my LCD voltages are in my M0517 charger - (in "calibrate"->"voltage" menu, battery disconnected) result:  0.005V. With 3cell battery connected the voltage on cell 4 was 0.065V 
(a little bit higher  but yours are 10x higher)

picture: 3 cell battery, MUX pin 3
[image]

interestingly I have only 9 voltage  "changes" per full cycle (not taking into account the small 
discharge fazes)  and your output has 17 (?),
It looks like your piking up also the "interlaced" measurements (shouldn't be possible (??)).
https://github.com/stawel/cheali-charger/blob/master/src/hardware/nuvoton-M0517/generic/50W/AnalogInputsADC.cpp#L94-L111

could you also connect a 3 cell battery (balance wire and negative main wire) and show me your oscilloscope output on MUX pin 3?

could you also measure the voltage on AVDD, AVSS pins on M0517 (pin 6, pin 42)
[image]

> If you look at 3 and 4 cell, there is also a 680mV after last cell.
hm.. interesting... but the 5 and 6'th cell is still 0V.

site note: 
your quartz is 12MHz (if I see it correctly) and the firmware assumes 16MHz (needs to by fixed in the future)

demianbadin, 2019-06-07:
> could you also connect a 3 cell battery (balance wire and negative main wire) and show me your oscilloscope output on MUX pin 3?

Picture: 3 cell & battery -
[image]

> could you also measure the voltage on AVDD, AVSS pins on M0517 (pin 6, pin 42)
Pin 6 = 0V
Pin 42 = 2.487V

demianbadin, 2019-06-07:
I've pulled out the resistor between MUX output and CPU AIn (1). And here is something interesting...

Picture 1 MUX side:
[image]

Picture 2 CPU AIn side:
[image]

stawel, 2019-06-07:
hm... there is clearly a bump on each measurement, this is definitely a bug in the code
(or M0517 is broken, but in my opinion this is very unlikely)

some paint "art", please ignore ;) 
[image]

ok... I have another idea, maybe this code works (in my charger) because I'm using a different compiler then you are, I'm using an old gcc:
```
 $ /usr/lib/ccache/arm-none-eabi-g++ --version
arm-none-eabi-g++ (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

could you try this firmware, and check if the bumps disappear (during 3cell battery measurement with resistor :) ) 
[targets.tar.gz](https://github.com/stawel/cheali-charger/files/3267789/targets.tar.gz)
(I was not sure which target you are using, so I've compiled all of them)
It's your master branch compiled with my compiler.

One more thing, tomorrow I'm going for 3 weeks to Korea and I wont have any computer with me, besides a tablet, so You will be on your own :/ , I can only support you with ideas :-)

demianbadin, 2019-06-08:
Thank you for the support. Tomorrow I'll test your file.
I'm using "cheali-charger-imax B6-50W-BOA6" target.
Have a good trip! (Without computer?? = S)

demianbadin, 2019-06-08:
I've tested your compiled file. Same result.
I'ld like to disable parts of the program. For example, puting a return in outputPWM function, until find the source of this pulses. It's possible?
Any suggestion?

demianbadin, 2019-06-10:
I found a point inside AnalogsInputsADC.cpp where the analog input pin1 rises to 3V.
Original lines:
void setADC(uint8_t pin) {
	ADC_SET_INPUT_CHANNEL(ADC, 1 << IO::getADCChannel(pin));
}

If I break the execution like this:
void setADC(uint8_t pin) {
        while(1) {}
	ADC_SET_INPUT_CHANNEL(ADC, 1 << IO::getADCChannel(pin));
}
the result is Picture 1:
[image]

If I break the execution like this:
void setADC(uint8_t pin) {
	ADC_SET_INPUT_CHANNEL(ADC, 1 << IO::getADCChannel(pin));
        while(1) {}
}
the result is Picture 2:
[image]

demianbadin, 2019-06-10:
Here is another thing in AnalogInpusADC.cpp lines 194-195:

Original:

>     /* Set the ADC operation mode as burst, input mode as single-end and enable the analog input channel 2 */
>     ADC_Open(ADC, ADC_ADCR_DIFFEN_SINGLE_END, ADC_ADCR_ADMD_BURST, 0x1 << 2);

Here you set the AIN2 for Burst read. In my case is AIN5.
It could be changed to?

>     /* Set the ADC operation mode as burst, input mode as single-end and enable the analog input channel 2 */
>     ADC_Open(ADC, ADC_ADCR_DIFFEN_SINGLE_END, ADC_ADCR_ADMD_BURST, 1 << IO::getADCChannel(MUX0_Z_D_PIN));

demianbadin, 2019-06-11:
Hi !, I have an advance.

I think the issue is in the initialization of the analog pin.
I've make the input work without problems, changing some lines in the initialization routine. And disabling the discharge operation.

1) Disable Discharge operation:

> void setMuxAddressAndDischarge (int8_t address)
> {
> if (address <0)
> return;
> g_muxAddress = address;
> ATOMIC_BLOCK (ATOMIC_RESTORESTATE) {
> // start mux ADC C discharge
> _setMuxAddress (ADC_CAPACITOR_DISCHARGE_ADDRESS);
> // IO :: disableFuncADC (IO :: getADCChannel (MUX0_Z_D_PIN2));
>}
> TIMER_Start (TIMER1); / * Start counting * /
>}
> extern "C" {
> void TMR1_IRQHandler (void)
> {
> / * Clear Timer1 time-out interrupt flag * /
> TIMER_ClearIntFlag (TIMER1);
>
> ATOMIC_BLOCK (ATOMIC_RESTORESTATE) {
> _setMuxAddress (g_muxAddress);
> // stop mux ADC C discharge
> // IO :: enableFuncADC (IO :: getADCChannel (MUX0_Z_D_PIN2));
>}
>}
>} // extern "C"

2) Change initialization of pin1 AIN5:

> // IO :: pinMode (MUX0_Z_D_PIN, ANALOG_INPUT); // _ DISCHARGE);
> // IO :: digitalWrite (MUX0_Z_D_PIN2, 0);
> ///////////////////////////////////////////////// //
> / * Disable the P1.5 & P1.7 digital input path to avoid the leakage current * /
> GPIO_DISABLE_DIGITAL_PATH (P1, 0x5);
>
> / * Configure the P1.0 - P1.3 ADC analog input pins * /
> SYS-> P1_MFP & = ~ (SYS_MFP_P15_Msk | SYS_MFP_P17_Msk);
> SYS-> P1_MFP | = SYS_MFP_P15_AIN5 | SYS_MFP_P17_AIN7;
> ///////////////////////////////////////////////// //
NOTE: my charger uses P1.5 (AIN5), but pin3 P1.7 (AIN7) is a free pin I've used to test purposes.

Also I've changed this line to make reference to MUX0_Z_D_PIN and not to pin "2".

> / * Set the ADC operation mode as, input mode as single-end and enable the analog input channel x * /
> ADC_Open (ADC, ADC_ADCR_DIFFEN_SINGLE_END, ADC_ADCR_ADMD_BURST, 0x1 << IO :: getADCChannel (MUX0_Z_D_PIN));

With this changes in AnalogInputsADC.cpp I was able to calibrate and get 0V when no battery is connected. I'll continue testing after work ...

Picture: 6 cell battery:

[image]

demianbadin, 2019-06-12:
So, after several tries, my conclusion is that I have no idea what appens.
I've tested 2 analog inputs, with same initialization, and same code, just changing pinout in imaxB6-pins, and the two AIns have different results.
AIN5 (pin1) has square pulses signal (outputs from charger) that seems to have relation with the execution of ADC_START. Picture 1

AIN7 (pin3) has a curved signal, like a charging / discharging capacitor. Picture 2

And the MUX output, disconected from CPU, is a razonable signal in relation to the cell inputs. Picture 3

I've flashed a second unit, same results.
I've purchased a Nuvoton replacement to test, M0516LDN. Hope it comes ASAP.
My head is burned out.

Picture 1
[image]
Picture 2
[image]
Picture 3
[image]