stawel/cheali-charger #275: Imax B6 original

JuStAdsor, 2019-06-01:
Hi!

I've tried to flash an imax B6 original using a USBASP interface.
I have first made the backup, and when asked if I want to restore the fuses I gave yes.
I have after flashed the latest firmware and answered Yes again. Tried the charger, but only squares showed up :(
Therefore I have decided to try again, but with the same results.
Once agan then I have tried another flash, but when asked about the fuses I gave no.
Now I can not connect anymore to it :(

Do you have any solution?
If I will remove the atmega32 chip and put another one, shoul I be able to restore it to the initial firmware?

What did I do wrong in the first place that it didn't work?

One thing to note is that in the first place I was powering the board from USBASP with 3.3v (was able to read the chip), but the display did not show anything on it. It was only lighten up.

Thank you!
Adi

diggit, 2019-06-01:
Hi,
do you have logs from flashing?
- It is good idea to lower programmer frequency ( `-D 100` for 10kHz clock), otherwise improper wiring may cause glitches in transfer.
- Charger is powered from 5V and not 3V3.
- when you dump original firmware, you have to look inside that hex file, if you see rising sequence, flash is locked and you can't extract original FW
- when avrdude asks about fuse correction, usually something is wrong (too high clock rate, bad wiring,...)

JuStAdsor, 2019-06-01:
Hi!
I do not have any log files.
Only the dumped firmware, but judging from what you say it might be corupted?
I can not connect to it anymore :(

flute2k3, 2019-06-21:
both the USBASP and B6 borad are not easy to damage, the most possible reason is your Uc locked, you have to use external clock to unbrick it. you can use any frequency source with 1Mhz or higher connect to Xtal1 or Xtal2 leg on B6 board. I successfully unbick mine.

JuStAdsor, 2019-06-21:
Hi!
I have just fixed mine last night.
Bought a new avr other than bother to unlock it as they are very cheap.
Resoldered the old one. Soldered a new one.
Flashed the new chip with custom firmware (it only worked after I have manually set the Christal frecvency in the command line)
After that it did not start since the fuses were wrong, so I have changed the fuses value according to the troubleshooting guide.
It started flawless :)

Thank you!

xpertcrush, 2020-07-04:
> Hi!
> I have just fixed mine last night.
> Bought a new avr other than bother to unlock it as they are very cheap.
> Resoldered the old one. Soldered a new one.
> Flashed the new chip with custom firmware [...]

kindly help a fellow user too, what did you do? i am having the boxes on screen.

JuStAdsor, 2020-07-04:
> kindly help a fellow user too, what did you do? i am having the boxes on screen.

Hi!

I have just replaced the MCU with a new one since the fuses seems too hard to reset and a new one is cheap.
After that I have just flashed it.

RadioControl, 2023-05-05:
> Hi! I have just fixed mine last night. Bought a new avr other than bother to unlock it as they are very cheap. Resoldered the old one. Soldered a new one. Flashed the new chip with custom firmware [...]

Hi, what frequency did You set? Do have pics or vid of the chip swap soldering? Thanks in advance!...