stawel/cheali-charger #276: LED =/!= PowerSupply?

emwu88, 2019-06-28:
How to use LED program?
Can it use like a standard PowerSupply?
When I set LED program, and run CHARGE, I get alarm "battery disc."?
With connected battery It works, without - no.
How to use LED program like a normal power supply (out voltage with 'open wire')?

diggit, 2019-06-29:
LED program works as constant current mode power supply. It requires load on output. Suitable for LEDs as name suggests.
Cheali does not have PSU mode. AFAIK there are limitations in HW preventing being decent PSU. Regulators are pretty lazy too (could be tuned), but that's not problem with batteries and such.

DaniLecx, 2021-08-03:
I've been trying to use LED mode on my Imax B6 too but I always get an error.
I have connected a led and set up the LED mode to 1.9V and 50mA.
When I start, the LED lights up then a second after the error "V limit" pops up.
Any ideas how to fix this behaviour?