stawel/cheali-charger #277: Compatibility with "Imax B6 LiPro Balance Charger 80W"

fiechr, 2019-07-13:
I got this device from China today.

Here are some pictures:
https://imgur.com/a/7j4pIHI

I'm not sure if the device is still compatible with the custom firmware.

The display is soldered over the MCU, but I can see a Atmel logo and avrdude tells me, it's a "M32" MCU (which is an internal alias for ATmega32). Also avrdude creates a backup of the stock firmware without any errors.
The pinout of the ISP header is the same as the original imax B6.

The device is sold as a "80W" charger, but as far as I can tell the binaries provided for download for the "original" B6 don't include a 80W version. So I'm not sure which binary to chose?

The PCB seems pretty recent (2019), although most changes look like they only switched from through hole to SMD parts.

Maybe someone else has already tried this?

brandon3055, 2019-07-13:
It looks like they have just taken the standard 50W design and added some beefier components.
If that is all that is different then the firmware for the original i6 should work. 

If this works you would be software limited to 50 watts. But if you can figure out how to compile the source it looks like you can easily up the power limit to 80w by tweaking the hardware config to 6amps and 80watts
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/imaxB6-original/HardwareConfig.h#L26-L27

I recently flashed two i6 50w clones. The boards looked much more like the original than a clone so i tried the firmware for the original i6 and they work just fine. Calibration was way out but thats easily fixed.

Now take everything i just said wich a giant grain of salt because flashing 2 chargers does not make me an expert. I would recommend waiting for some input from an actual expert on the matter.

Good luck.

diggit, 2019-07-13:
According to images, there is additional IC in top right corner, probably op amp. It is usually differential amplifier for all bottom 2 balancer channels. (usually 2 lowest cells have simplified circuitry in those cheap chargers) I would try [B6AC-80W](https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/Dual-Power-B6AC-80W-RC/HardwareConfig.h) which seems to match your version. No guaranties though...
Use hex files from "unstable" directory, they are most recent and pretty stable.

brandon3055, 2019-07-14:
@fiechr Would it be possible for you to post the part numbers for the fets and diodes used in your charger? I kinda want to experiment with the idea of upgrading a 50w charger now.  
The parts i am interested in are around and possibly under the bottom left corner of the LCD.

fiechr, 2019-07-14:
I have added an image with the components I could identifiy/read (all ICs). I did not remove the LCD, but the arrows point to the position under the LCD.

Not related to the software: Unfortunately my device seems already broken (using stock firmware). First I got a high current on the input and the display didn't turned on, at the next try it worked fine for another 20 minutes and after that I got some smoke in the vicinity of the input capacitor after trying to turn it on again. I think I will try to remove the cap. But the device labeled "4407 GA1821" also looks fishy too me. Now even my power supply shuts itself down when connected to the charger.

brandon3055, 2019-07-14:
May be a bad SMD capacitor. Possible one of the three directly bellow the input jack.

Those components are rather interesting. The 4407 is the same as mine. Its a P-Channel fet rated at 12 amps so it would responsible for the buck side of the charger's buck-boost regulator. 

Those diodes seem to be s significant downgrade compaired to mine as they are only rasted 1 amp as opposed to 5 amp in my unit. If i had to guess whats going to die first it would be those because one of them is going to be seeing 6 amps continuous at max load. 

The NCE40H12K is the most interesting component. My charger uses a fet rated for 10 amps in that position but the NCE40H12K is rated for 120 AMPS! But based oon ebay prices it looks like its about the same price so i guess they figured why not?

Not sure if you find any of that as interesting as i do but there it it.

fiechr, 2019-07-14:
I'm glad if I could contribute some information on the hardware part.

The smoke is coming from the 4407. I've checked the SMD caps, but they all seem fine.
I guess I'm trying to get a refund.
Not sure if it would make sense to replace the FET or if the real issue is somewhere else.

brandon3055, 2019-07-14:
It seems unlikely that fet would be doa. There is probably some other factor that contributed to it.
If you feel like messing around with it you could try preplacing it with a similar or identical fet but if you just need it to work then your best bet is to get a replacement.

dischinator, 2019-08-15:
@fiechr did you fix your charger and try a cheali build? From the looks of your photos I got the same charger, except mine has no PH soldered on, and I'd like to flash it to get lipo hv support.

fiechr, 2019-08-15:
Hi,
I'm not sure, if I really fixed it. I replaced the MOSFET and tried to flash different downloadable binaries of the Cheali firmware. Unfortunately, that didn't work (charger didn't start up, showed only a blank line), so I decided to rebuild everything from source. I also created a new target with the following settings:

```
#define MAX_CHARGE_V            ANALOG_VOLT(27.000)
#define MAX_CHARGE_I            ANALOG_AMP(6.000)
#define MAX_CHARGE_P            ANALOG_WATT(80.000)

#define MAX_DISCHARGE_P         ANALOG_WATT(10.000)
#define MAX_DISCHARGE_I         ANALOG_AMP(2.000)

#define SETTINGS_MAX_DISCHARGE_I        ANALOG_AMP(0.300)
```

The resulting binary works fine. I don't know, what could have changed, maybe it had other reasons not work at first.

I'm also pretty sure, that it wasn't necessary to create a new target for these chargers, since these are all more or less "soft" values, that only make sure, that the charger isn't overloaded. To be honest, I later reduced most of the values in the firmware settings to more conservative values, since I don't trust the device can really handle, what it promises on the package/case.

The only real difference seems to be this line:
`#define SETTINGS_MAX_DISCHARGE_I        ANALOG_AMP(0.300)`

But I don't know what it really does. It's part of the "imaxB6-clone" target, but not of the "imaxB6-original". Maybe the latter one uses the default value of this variable.

Since I got another charger in the meantime, I haven't really used it since then. So I can't tell if everything works, but it looks promising. Calibration also seemed to work fine.

dischinator, 2019-08-16:
@fiechr thanks for the detailed answer. Almost sounds like the build from source itself made it work for you. I'll try with your settings. But in the long run buying a new charger also sounds like a good idea since the clone one really is a bit sketchy and no one wants a burning lipo.

dischinator, 2019-08-17:
@fiechr update: as a first test I build from source and flashed target Dual-Power-B6AC-80W-RC since @diggit mentioned it. This target instantly worked, calibration also seemed fine.

fiechr, 2019-08-18:
That seems also to confirm what I presumed (that creating an additional target wasn't necessary).

Did you try the ready built binaries first or did you go straight to compiling? Would be interesting to know - since I can't really think of a reason, why the provided binaries shouldn't work exactly the same way.

dischinator, 2019-08-18:
No I did not try the prebuilt binaries. Maybe I'll backup and try them later.

fiechr, 2019-09-08:
Today I finally calibrated the charger and charged an actual battery with it. I can confirm, that the firmware works fine with it. I still don't trust it to handle the values it was advertised with, but normally I charge with much lower currents anyway.

dischinator, 2019-09-09:
Good to know. I'm charging my 3" copter batteries with it charging power is below 30W should be okay. so far no problems at least.

sklepa, 2019-09-30:
I have the same device, I've flashed it succesfully, and everything works, besides two first cells are "floating" around. No matter how many times i compensate their voltage in calibration - they end up being wrong. On the other side further cells i.e. cell3 and cell4 keeps their calibration spot on. I've been trying to make expert calibration, but i cannot force the charger to show up voltage on pin0 of the balancer, and dont know why. The resistance between pin0 and GND is sufficient high. Any ideas what to make to improve those 2 first cell readings ?

diggit, 2019-10-04:
Hi @sklepa,
which FW file did you flash (full name)?
What resistance do you measure between pin0 and GND?

sklepa, 2019-10-11:
Hey @diggit, i flashed this one:
cheali-charger-imaxB6-clone_2.01-e10.3.12-20170311_atmega32.hex

and resistance measured is exactly: 5,55 Mohms  
isn't it a bit much ? i am pretty convinced that measurment was proper, i have used 2 different fluke multimeters, and was probing couple ground points on whole board.

diggit, 2019-10-11:
Regular imaxB6 has simplified circuit for bottom 2 cells measurement. These 80W versions have diff amps on all channels. Try `cheali-charger-Dual-Power-B6AC-80W-RC_*_atmega32.hex` which expects diff amps on all cells. Builds in "Unstable" folder are fine.

sklepa, 2019-10-11:
Thanks! I'll try it and report the results!

sklepa, 2019-10-11:
@diggit Damn! it's working perfect right now! +-1mV of error. Sweet!
Thank You. And if I may :) another question: How to change charge/discharge software limits ? are they in eeprom or should i compile a new modified hex ? and where to find some info? I looked in repository, and there is a "building from source" but i don't quite understand where to start.

diggit, 2019-10-21:
Those limits are hard-coded in FW. Unless you modify HW, you should not change(increase) them.

fiechr, 2019-12-05:
> Regular imaxB6 has simplified circuit for bottom 2 cells measurement. These 80W versions have diff amps on all channels. Try `cheali-charger-Dual-Power-B6AC-80W-RC_*_atmega32.hex` which expects diff amps on all cells. Builds in "Unstable" folder are fine.

Interesting! How can I tell the difference by looking at the boards?

This is the clone charger I have: https://imgur.com/a/7j4pIHI

Is it the LM2904 on the top right corner?

I guess the "ENABLE_SIMPLIFIED_VB0_VB2_CIRCUIT" flag switches between one and the other.

diggit, 2019-12-05:
You can tell by IC presence (SOIC8 dual op amp) near ballancer (top right mounting hole). If it is not there, you have simplified version which is more common within 50W chargers.

gregcrago, 2020-01-16:
I have same 80W charger.
Why is #define SETTINGS_MAX_DISCHARGE_I ANALOG_AMP(0.300)
and not set to #define SETTINGS_MAX_DISCHARGE_I ANALOG_AMP(2.000) ??
It is rated for 2A discharge

diggit, 2020-01-16:
Where did you find it?
[B6AC has 2A discharge limit](https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/Dual-Power-B6AC-80W-RC/HardwareConfig.h#L34).

gregcrago, 2020-01-16:
found it here:
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/targets/imaxB6-clone/HardwareConfig.h

diggit, 2020-01-17:
That is config for Imax B6 clone (never met this type personally). Yes, it is suspiciously low, but this issue is about 80W version which uses B6AC config. (I linked earlier, witg Idchg=2A limit)

LynxChaus, 2020-02-10:
Guys, you believe in smd 2512 shunt will survive when discharging full 6S cell battery at 2A? Really?
I have clone with board 2019.10.21 B6-80W v:2.2 with same smd 2512 shunts.

dischinator, 2020-02-11:
@LynxChaus theres also a power limitation for charging/discharging in place, for example for the 80W charger:

#define MAX_CHARGE_P            ANALOG_WATT(80.000)
#define MAX_DISCHARGE_P         ANALOG_WATT(10.000)

It will discharge at max 10W i.e. about 600 mA at 4s, which really takes a long time and the charger still gets very hot since it has no fan. It's not really a good charger for discharging.