stawel/cheali-charger #282: Do PB settings suite AGM batteries?

hryamzik, 2019-12-06:
I've replaced my PB battery with an AGM one and was warned that regular PB charges can damage it.

I have to say that sounds ridicules as these batteries are supposed to suite regular cars with the same charging parameters as PB batteries had.

The only limits I managed to find are max voltage of 14.4 V and max current 10% of capacity.

diggit, 2019-12-07:
That should be it. 2.4V per cell should prevent gassing which AGMs don't like. Also setting reasonable charging current is good idea.

hryamzik, 2019-12-09:
Thanks, that's exactly what I've done (2,4V, 5A for 60A/H). I've seen voltage temporary getting higher than 14.4 on the last seconds of charging, hope it's not that bad.

closed 2019-12-09 by hryamzik