stawel/cheali-charger #286: "Imax B6 LiPro Balance Charger 80W"

laurentdrone, 2020-02-24:
hello i bought this one on the web
[](https://drive.google.com/drive/folders/1QxuVCSdyOpc44TmpPcbGmIu6z-I-_5di)
imax b6 lipro balance charger
I do not find it in the doc.
on the top of the pcb I have 7 pins instead of 6
Someone already seen this one
thank you

Johnnysod, 2020-02-27:
I just got one like that. Not great that there is no print on the chip.
Could be a Nuvoton-M0517
All I've found so far is the interface pins are
5v-0v-??-??/5v-0v-??
Can you see/test which pins on the chip go to the ?? on the interface ...
There was a bit of voltage (2v) on the right hand ?? pin.
We could then work out what they are and maybe which chip we have!
I'm thinking of sending mine back if I can't flash it as I need a LiPo HV mode.

laurentdrone, 2020-02-28:
thanks for the reply.
I will see with the seller for the MCU
me too i need the lipo HV
i will be watching this weekend

Johnnysod, 2020-02-29:
However the board in the advert picture (brown) looks like my other charger and not like these new green ones!

diggit, 2020-02-29:
Well, I see 5V regulator only. 5V ARM MCUs are quite rare (not cheap). I *guess*, It's some 8bit MCU, maybe 8051 based. Because they are cheap.

laurentdrone, 2020-02-29:
I asked the seller to inquire here is his answer

Hello, I'm very sorry, the manufacturer's microcontroller parts are classified as manufacturing trade secrets, and the manufacturer did not disclose this information. Therefore we cannot reply to your 

loll

XETERRR, 2020-03-21:
Hello!

I also have that aliexpress charger! But Default calibrate amperes bad, it would be nice to update firmware!

[image]
[image]
[image]
[image]

I am looking forward to solving the problem. Tranks.

alin1337, 2020-04-17:
Hi, i got the same model (was really cheap, < 20$).
I unsoldered the display to take a look at the cpu and it seems its unmarked (no text on it).

xromansx, 2020-07-25:
Did someone find what's the CPU?

xromansx, 2020-07-25:
Some internal photos:
[image]
[image]

carlosfd1, 2020-10-24:
Hello!
Has anyone figured out what chip it is and what the pinout is?

medbenmedben, 2020-12-09:
I have the exact same model, I'm investigating.

beli11, 2021-04-17:
I have the same model. The CPU is an "ABOV MC96F6432Q 1951". I could trace the schematics if someone is interested in adapting the code!

MkeFobe, 2021-08-13:
I've got the same charger. Thanks @beli11 for investigating the MCU!
Unfortunately it one of that crapy 8051 with too less memory :-(
I'll have to live with it. Not worth to invest time on rewriting the code...
Life is too short for bad MCU's ;-)

MultiAnything, 2021-08-23:
got the same version of board / unmarked cpu, did somebody already have compared the schematics / board traces...
thought about what happenes if i try to replace this unmarked CPU with an atmega32x

by the way: it's an cheap 20bucks charger, would be nice to bring this parts together and have an unbugy software version to get an relieable charger for this mony would be awesome

regards

diggit, 2021-10-13:
It's N-channel mosfet transitor used in boost part of smps.

pitfok86, 2021-10-30:
Hey, after replacing the 470uf 25v capacitor and the IRF 7413 transistor, the charger turned on. But squares appeared on the display and nothing else could be done: / Anyone have an idea?[

pitfok86, 2021-10-30:
> @pitfok86 @keskin10 how did you manage to destroying it? Was it with stock FW or cheali?

Hi. I bought a damaged one for repair, the previous owner mistook the poles at the dc input and hence such damage. Now, after the exchange, further questions arise, but this is what I wrote on Github

pitfok86, 2021-10-31:
hey, anyone knows what is this smd resistor?[image]

[image]

Johnnysod, 2021-10-31:
> hey, anyone knows what is this smd resistor?[image:
> 88447356-559e9700-ce00-11ea-812f-120a9ca52e5e~8.jpg]
> <https://user-images.githubusercontent.com/83096138/139585039-d915247c-622a-4d92-b855-25d8b723d4d1.jpg>
>
> [image: IMG_20211031_090513865~2.jpg]
> <https://user-images.githubusercontent.com/83096138/139585047-bad122af-bc67-4c14-8195-03ccafb44b43.jpg>

Does this help

John Dyson
07764 335343

Johnnysod, 2021-10-31:
I think the resistor is in series with the 5v REG and doesn't exist on the
BC6 diagram. It looks like it says 100 but  I'm not sure if that means 10
ohms

John Dyson
07764 335343

MultiAnything, 2021-10-31:
Hi, R19 is 10 Ohm meassured and seems to be the value printed on the body
(SMD size 1206 should possible fit :) )

good luck and nice Happy Halloween

pitfok86, 2021-10-31:
> Hi, R19 is 10 Ohm meassured and seems to be the value printed on the body
> (SMD size 1206 should possible fit :) )
> 
> good luck and nice Happy Halloween
> 
> 
> > I think the resistor is in series with the 5v REG and doesn't exist on the
> > BC6 diagram. It looks like it says 100 but I'm not sure if that means 10
> > ohms
> >
> > John Dyson
> > 07764 335343
> >
> >
> > > Does this help
> > >
> > >
> > >
> > > John Dyson
> > > 07764 335343
> > >
> > >
> > >> hey, anyone knows what is this smd resistor?[image:
> > >> 88447356-559e9700-ce00-11ea-812f-120a9ca52e5e~8.jpg]
> > >> <
> > https://user-images.githubusercontent.com/83096138/139585039-d915247c-622a-4d92-b855-25d8b723d4d1.jpg
> > >
> > >>
> > >> [image: IMG_20211031_090513865~2.jpg]
> > >> <
> > https://user-images.githubusercontent.com/83096138/139585047-bad122af-bc67-4c14-8195-03ccafb44b43.jpg
> > >
> > >>
> > >> <
> > https://github.com/stawel/cheali-charger/issues/286#issuecomment-955697766
> > >,
> > >> <
> > >
> > >> <
> > >
> > >> or Android
> > >> <
> > >>
> > >>
> > >
> >
> > or Android

hey, thank you for the answer, I will try 😉 because I'm curious myself 😬😁 if anyone else has suggestions, please write. Thanks in advance and wish you a Halloween 😁 too

keskin10, 2021-11-01:
@pitfok86 

> hey, bu entegre devrenin ne olduğunu bilen var mı?
> 
> [image]
> 
> [image]

@pitfok86 the element here is not IRF7413.

keskin10, 2021-11-01:
[image]

diggit, 2021-11-01:
Damn, I did not notice, there is space between that transistor and DC- polygon. As it is not in boost, but it's buck part of smps, IRF7413 won't do the job, because it's Nmos and you need Pmos. You can use NCE4435 as @keskin10 shows or some equivalent like AO4409 and many similar.

Sorry for that @pitfok86

k1r1t0-sti, 2022-06-15:
Hello everyone, this model has a critical problem, after a successful full charge, the charger consumes energy from the battery, with a current of about 100 mAh. If you exit the charging mode, the consumption will stop. If you leave the charging process unattended, the battery will be discharged.

Johnnysod, 2022-06-15:
I've not noticed mine doing that with the cheali firmware.
Sounds like the discharge circuit is active for some reason..

John

k1r1t0-sti, 2022-06-16:
This is manifested on the stock firmware, unfortunately there is no cheali firmware for this processor. (Zilog 8051 Z51F3220FNX)

Morad-T, 2024-10-07:
> Some internal photos: [image] [image]

I'm really sorry to revive a dead thread. I have been searching for the schematic of the variant pictured above by @xromansx and @pitfok86 
If you happen to have it I would be eternally grateful. I am trying to repair mine but unfortuanately the burnt part is unrecognizable.
[image]
It is the transistor/IC just beside the enter button.

Thank you in advance

Johnnysod, 2024-10-23:
> Hope this helps

John Dyson
07764 335343

Johnnysod, 2024-10-23:
Sorry for the late reply

John Dyson
07764 335343

Morad-T, 2024-10-28:
> Hope this helps [...]

I can't thank you enough man. I really really appreciate your effort.