stawel/cheali-charger #287: Cheali FW for Turnigy-Accucel-6-50W-6A is working only with RST signal

vladT0, 2020-03-03:
Hi !
I successfully flashed Cheali FW for Turnigy-Accucel-6-50W-6A.  New firmware starts only if I connect 
VCC, GND and RST from USB, in other case it is just hang. With just normal power supply attached I have  squares at the display. Tried to reset fuses according your recommendation. Did not help. Tried
to reset eeprom from the settings, also did not help.
Thanks in advance for your help
Vlad.

vladT0, 2020-03-03:
Log file for the prev. post.
```
>avrdude.exe -patmega32 -cstk500 -Uflash:w:cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9502 (probably m32)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex"
avrdude.exe: writing flash (30810 bytes):

Writing | ################################################## | 100% 7.81s

avrdude.exe: 30810 bytes of flash written
avrdude.exe: verifying flash memory against cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex:
avrdude.exe: load data flash data from input file cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex:
avrdude.exe: input file cheali-charger-Turnigy-Accucel-6-50W-6A_2.01-e10.3.12-20170311_atmega32.hex contains 30810 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 6.42s

avrdude.exe: verifying ...
avrdude.exe: 30810 bytes of flash verified

avrdude.exe: safemode: Fuses OK (E:FF, H:C5, L:3F)

```

stawel, 2020-03-04:
I'm not sure what is happening there, but:
measure the voltage on VCC, RST (according to GND) on your ISP connector with and without the programmer connected,  you should get VCC=~5V, RST=~5V in both cases.
If your chargers MCU is powered with a lower voltage (for example 3.3V if programmer is not connected), try to change the BODLEVEL bit to 2.7V:
`avrdude.exe -patmega32 -cstk500 -U lfuse:w:0xbf:m -U hfuse:w:0xc5:m`

**Important!**
make sure I've calculated the fuses bits correctly, I tend to mess those things up :/
go to:
1. http://www.engbedded.com/fusecalc/
2. in "device selections"  select ATmega32
3. in "current settings" enter 3F, C5 and press "Apply Values"
4. change "Brown-out detection level at VCC=4V" to "Brown-out detection level at VCC=2.7V"
5. you should see the new fuse values in "current settings"
6. **please do not change any other settings, or you might brick you charger**

vladT0, 2020-03-04:
Thanks a lot Pawel for the quick response. I will check it today and let you know about results.  Short question about fuses.  I noticed in your recommendation you have made changes in BOOTSZ1 and this way extend application area in flash and made smaller boot-loader area. I did not investigate in details your source code, just idea. Is it possible with wrong fuse,  with a simple goto to jump from bootloader to the wrong start of the application ? but this idea can not explain, why it is working with RST pin powered from programmer.

vladT0, 2020-03-04:
We can close it. It was defect of PCB near RST pin.

closed 2020-03-04 by vladT0