stawel/cheali-charger #288: Parallel charging and set capacity

fiechr, 2020-04-05:
I was just parallel charging two 550 mAh 1S batteries and the overcapacity feature stopped the charging - which makes sense, since it's 2x the set capacity.

Simple question: Is there any way to ignore the capacity in the charging process? The minimum value for this settings is 1% and the maximum 250%.

If the number of batteries may change, so the only idea I have is to set the capacity to the maximum value.... but then again, if I charge larger batteries in parallel it may be not high enough.

fiechr, 2020-04-05:
Ok, ignore the question, please!

I found out, that the capacity can be set to "no limit".

diggit, 2020-09-25:
In cheali philosophy, you should have separate configurations for each battery or combination, but your solution works too. Parallel charging is not safest thing, but I guess you know the risks...
If you feel, that issues is solved, please close it.