stawel/cheali-charger #289: Trickle charge funcitonality

Iscle, 2020-04-10:
Now that coronavirus is around, lots of cars are stopped and their batteries can get low. To keep them from dying, it would be nice to have a trickle-charge feature for Pb batteries.

I'm looking into how to implement this by myself, but it would be nice if someone with more experience with this firmware could do it (a lot faster than me).

Thanks!

diggit, 2020-09-25:
Hi, AFAIK trickle charging is charging by short pulses. This is generally not easy to accomplish, because these chargers has large capacities on their outputs.

Iscle, 2020-09-25:
You're right.

What about a mode to keep them between a certain range? For example, letting them discharge until they hit 11.5v and then charge them back up to the required full-charge voltage.