stawel/cheali-charger #291: eeprom error: 2 yes after enabled power

githubniko, 2020-05-17:
Hi,

Everything works well, but sometimes when the power is turned on, the error "eeprom error: 2 yes" pops up, after which you have to re-calibrate the device. Tell me where can I fix this?

Все работает хорошо, но иногда при включении питания выскакивает ошибка "eeprom error: 2 yes", после чего приходится заново калибровать устройство. Подскажите где это можно исправить?

githubniko, 2020-05-17:
v2.0 and 2.01
cheali-charger-imaxB6-original_2.00-e10.3.12-20160613_atmega32.hex

gumileeev, 2023-03-19:
I have the same problem on turnigy accucell 6a 2.0 and 2.01
eeprom error: 4 yes

githubniko, 2023-03-19:
As I understand it, the problem occurs if there is a lack of power during on/off.

stawel, 2023-03-20:
This error means that we checked the _eeprom_ (where all permanent data is stored) and its corresponding _CRC_
and  we found an error, so some data is corrupted.

The meaning of the numbers can be found here:
https://github.com/stawel/cheali-charger/blob/master/src/core/eeprom.h#L28-L31

In case of @githubniko the calibration part is corrupted and in case of @gumileeev it's the program part (batteries charging settings).
After pressing: "yes", the corresponding part of the _eeprom_ will be reset to _default_.

### What could be the cause:
(this is just my speculations)
1. during _eeprom write_ there was a power shortage and the CRC was save incorrectly, **BUT** we only write to the _eeprom_:
    - during  calibration
    - when you change settings
    - when you change program data (batteries charging  settings)
2. there is a bug in the software. We only had a few such reports so  I'm not sure if this is likely, but I might be biased :)
3. the CPU is damaged - the _eeprom_ CRC check was introduced for this case
4. you have some issues with you power supply. If your power supply occasionally goes below  the required CPU voltage but not enough to reset the CPU, then the CPU might execute some code at random place - there is a chance that it hits a `eeprom::write` code
5. you are living  in a place with high radiation (probably unlikely)

In case of 4. and 5. you probably would have seen some weird behavior before this and you probably would get different error numbers each time (with some probabilities).

### What can be done:
* reset atmega32 fuses: see https://github.com/stawel/cheali-charger#troubleshooting
    - this might sometimes help with 4. since we set the _BODEN_, _BODLEVEL_ flags (Brown-out detection)

* resolder a new CPU - most likely this will help

#### Final thoughts
You might also check if the CPU is damaged by:
- set/calibrate everything you need to set/calibrate
- save the _eeprom_ to your PC with for example: _avrdude_ 
- wait for the error to reappear
- save _eeprom_ to your PC again

if there is a difference between  the saved _eeproms_ in only one bit (or one byte) then it is a CPU issue.
At this point you could also  write the first eeprom  back to the charger so you don't have to recalibrate, but I would resolder a new CPU.

gumileeev, 2023-03-20:
Thanks for your reply! 
I changed the fuses again, as required by the instructions, installed the program again, and no more errors. Thanks for your reply and your program! It will be great if you update the version one day and make a change in the sound volume for the turnigy a6 50w. I am very happy that everything is working well!

gumileeev, 2023-03-20:
> 

Thanks for your reply! 
I changed the fuses again, as required by the instructions, installed the program again, and no more errors. Thanks for your reply and your program! It will be great if you update the version one day and make a change in the sound volume for the turnigy a6 50w. I am very happy that everything is working well!

closed 2023-03-21 by githubniko