stawel/cheali-charger #292: MonsterTronic MT680AC

sharkyenergy, 2020-05-26:
Hello! 

is this charger supported? it has an atmega chip, and the layout looks very similar to the original B6, but is not identical. The original software is full of bugs..

https://monstertronic.org/details/E-04

thank you!

TBAMax, 2020-05-27:
Hello
Can you share pictures of the board?

sharkyenergy, 2020-05-27:
sure! here we go:

[image]
[image]
[image]
[image]

diggit, 2020-05-28:
At first glance, pinout seems same as other chargers. This one features additional OpAmp for proper measurement of bottom 2 cells (common for 80W charger variants). I'd try `cheali-charger-Dual-Power-B6AC-80W-RC`, but no warranties.

sharkyenergy, 2020-05-28:
huhm, can I still restore the current situation in case it does not work or will I brick it? the current original firmware stops automatically with alarm at 5000 ma... so I cannot fully charge a battery in one go.. and there is no way to change this setting.. what about the dual charger? will it work even if mine has only one charging port?

diggit, 2020-05-29:
You can also check pinout, [here is pin mapping used in cheali](https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/generic/50W/imaxB6-pins.h) and [here is schematic](http://www.rcgroups.com/forums/showatt.php?s=df7049bcbafdb5d7d06765c264e5c4bb&attachmentid=3693125&d=1293732709) (component references probably won't match).

Well, at first, you can try to make backup of [original FW](https://github.com/stawel/cheali-charger/blob/master/docs/flashing.md#flashing-using-avrdude). Once you have `hex` files, open `flash.hex` in plaintext editor and look if there is increasing sequence of hex numbers, if so, you MCU is protected against readout (highly probably). If not, you have obtained backup which can be flashed back any time.

All these chargers have one MCU per battery port, besides that, `dual-power` in that filename means, that charger could be powered from regular wall socket (AC) or some DC PSU. No effect on FW.

sharkyenergy, 2020-05-29:
thank you very much!! will try this weekend! :)

sharkyenergy, 2020-06-01:
It works like a charm! the code was not protected, so I still have a backup of the original one.. But this works wayyy better.. (although its slower in charging...) So i can confirm that the MonsterTronic MT680AC is compatible with the firmware cheali-charger-Dual-Power-B6AC-80W-RC

Thank you very much everybody

closed 2020-06-01 by sharkyenergy

diggit, 2020-06-01:
>  its slower in charging
how? lower max current/max power limits? That could be fixed...