stawel/cheali-charger #294: Why is it one graph?(LogView v2)

dualfumi, 2020-06-13:
Why is it one graph?

iMAX B6(original)
cheali-charger-imaxB6-original_2.01-e10.3.12-20170311_atmega32.hex
ChealiCharger_6cell_1.0.ini
[NIMH]-[D>C format]
LogView bUILD:V2.7.5.517
[image]

diggit, 2020-06-13:
Hi,
Isn't plot config matter of LogView? You should be able to change plot properties as much as you want (and LogView allows).

Also, please describe what behavior do you expect vs what happened, when submitting issues.

dualfumi, 2020-06-13:
Hi.
Thank you for your reply.

I am Japanese so I am not good at English.
I'll do my best, so thank you.

Since the language of logview is English/German, I still don't understand the detailed settings.
I'll look for more.

When cycled with imax b6+logview, separate discharge/charge graphs were obtained even with the standard settings.

When I cycled with cheali+logview, discharge/charge/discharge/charge became one graph, as shown above.

So I thought it was due to the difference in the operation processing in firmware or the ini for cheali.

Best regard.

diggit, 2020-06-13:
FW only transmits values. It's up to other SW how it presents those values.
There is nothing about plots in [ini file](https://github.com/stawel/cheali-charger/blob/master/utils/LogView/ChealiCharger_6cell_1.0.ini).
I am not using LogView so, I can't help with that. Maybe others...

dualfumi, 2020-06-13:
I am not concerned with logview. I would like to try if there is alternative software.
Can you tell me if you have the software you are using?

dualfumi, 2020-06-14:
The contents of ChealiCharger_6cell_1.0.ini.

[Stati]
StatiAnzahl=11
001=Charge
002=ChargeBalance
003=Balance
004=Discharge
005=FastCharge
006=Storage
007=StorageBalance
008=DischargeChargeCycle
009=CapacityCheck
010=EditBattery
011=Calibrate

I think it's the condition of the charger.
For cycle D/C, there is only one type of "Discharge Charge Cycle".

For example, if there are two kinds of states such as 008-1=DischargeChargeCycle(Discharge) and 008-2=DischargeChargeCycle(charge), I think that the graph was automatically divided.

However, to achieve this, I need to modify the firmware, and I do not have such a skill.

Best regards.