stawel/cheali-charger #295: Question about burned up SMD resistor while calibrating current

fiechr, 2020-06-16:
Since I'm very happy with the imax B6 (one original and one clone), I got another clone to install the Cheali firmware on.

While calibrating the current a high default current of about 1.5A was the starting point of the calibration and before I could unplug anything the left one of the large resistors burned up.

The only reason I'm writing this here is to ask, if this no surprise and totally "normal" and my fault alone, because I select too high default values when building the firmware. 

I now got two new clones and obviously don't want to burn them, too.

So I want to make sure to identify the reason first. The battery connected at the time was a 4S LiPo with about 15 V. 1.5A * 15V = 22,5W, which is quite a bit. I this enough to burn the resistor in no time or should they both (if in series) have handled that at least for a few seconds? I also wonder why only one burned while the other is still fine (maybe the weakest link...).

Maybe I also should use another, lower voltage battery after voltage calibration, just to reduce the risk and load.

Without any additional information, I was planing on reducing the default calibration values significantly before flashing and calibrating. But I just wanted to ask if this could also be partly due to a bad component?

I'm building it myself, since I want lower default values for discharge power and amps, without the possibility of changing it back in the settings, while keeping the higher settings for charging. Mainly to prevent exactly what happend accidentally.

[image]
[image]

diggit, 2020-06-16:
Hi,
Those are 2512 (package) resistors and it can handle up to 1W. There resistors are just current shunts. Main power loss during charging is on transistor. *Usually*, there is dedicated resistor for charging (red) and another one for dischaging (green)

[image]
*values may differ, this is partial schematic of 50W charger*

Do you know what was the value?

Discharging calibrates at lower currents (100 & 300mA?), because range is limited (1A @ 50W charegrs). What is discharge current rating of your charger?

diggit, 2020-06-16:
Charging shunt resistor should handle 5A pretty well. What was the real value of current on charger output? Did it smoke when you calibrated charging or discharging current?

fiechr, 2020-06-16:
The discharging was the problem.

I'm currently testing the new device and the value (since I don't remember the values of the broken one, but this one is identical) for

the 50 mA calibration point is 2610 in the first line and in the second line it's showing 16953.

and the 300 mA calibration point is 500 in the first line and in the second 16957.

This is after resetting the EEPROM on a never used device.

The first values correspond to the values set in the defaultCalibration.cpp on the last line

I did not dare to actually start the calibration of the discharge current, yet.

diggit, 2020-06-16:
I am not very familiar with values on display during calibration. So these are best guesses...
On top, the is probably PWM value, on bottom ADC value (oversampled). ADC value does not make sense when not calibrating.
In my (modified) charger, I have PWM value (first line) == 188 for 50mA range and 3170 for 1A. At least, PWM value for 50mA should be lower than for 300mA (thta 2610 is really weird). IDK what is happening there, but I'd try replacing default values by those from Imax B6 original.

Imax B6 clone has default calibration for Idischarge like this:
`{{7095, 50},                        {43070, 300}},    //Idischarge`
regular Imax B6 (imo more common) has:
`{{2017, 50},                        {11971, 300}},    //Idischarge`
those are 2 calibration points pairs `{ADC, current in mA}`

fiechr, 2020-06-16:
I think I know why there are these much lower values in the defaultCalibration.cpp -- because I changed them! I **thought** the problem was, that the start value was to high and so I changed them to much lower values. But this was a few weeks ago. I just wanted to make sure not to forget it in the end when trying again with a new device.

I know that I used the exact same build with the same default values for another clone a few months ago and never had any issues while calibrating or using it. So there is a good chance, that there was something wrong with the board.
Unfortunately,I don't know enough about electronics to find that out by myself. For example if the MOSFET above is faulty, could this happen?

Thank you for your time and help! :-)

fiechr, 2020-06-17:
This time the calibration went fine without any issues. The values match exactly the values measured with my multimeter. BUT unfortunately whenever I try to charge anything, I get "calib. error I charge 14". :-(
[image]

fiechr, 2020-06-17:
According to this page:
https://github.com/stawel/cheali-charger/blob/master/docs/calibration_error_codes.md

I measured and calculated the following values:

I charge:

100mA: value1 = 1282; ADC1 = 1282  --> -15.6601 (maximum); 5889.15 
1000mA: value2 = 11250; ADC2 = 11253 --> -15.6253; 5887.4 (minimum)

So I've set max Ic to 5900 mA (was on 6000 mA) and min Ic to 10 mA (?) - not sure about the negative value...

Charging now works! :-)

Then I did the same for the discharging current calibration and calculated the values -14.0883 and 10286,3
I'm not sure what to do with them, since the are obviously too high, so I did set them to 50mA and 500 mA. Did not test discharging yet.

The values for I discharge during calibration were:

value1: 1142; ADC1: 3139
value2: 3177; ADC2: 8640

diggit, 2020-09-25:
It's great, that you managed to fix your charger and calibrate it. Good job! If you think, that this issues is solved, you can close it.

closed 2020-10-02 by fiechr