stawel/cheali-charger #296: Is it possible to set the charge/discharge cycle beyond 5?

megamaniac, 2020-07-01:
If so, is it then possible to use this charger in combination with Logview to do 'cycle life testing', cycling the batteries up to hundreds of times and recording the data?

diggit, 2020-09-25:
At this moment, cycling is limited to 5 cycles, because results from each cycle are saved for review (charge in/out). Space for all of them has to be reserved at compile time at compile time (no heap). Resources in these MCUs are limited.

It shoudn't be complicated to modify FW to store eg. only last 5 cycles and allow more cycles in total.

TBAMax, 2022-12-25:
Seems like there I already memory allocated for 10 cycles in the ScreenCycle.cpp! It just is not used yet at the moment.