stawel/cheali-charger #300: Calibration charging current problem

38293312, 2020-10-12:
Hello there! I built a charger based on the hardware you provided. Now there is a problem that the charging current cannot be calibrated, and there is no charging current. The calibration discharge current is normal. If it is charging, it will prompt "calib.error voltage 4". I only used a 4S lithium battery for voltage calibration, and the battery is not in a charged state. Hope you can help me solve this problem, thank you!

diggit, 2020-10-13:
Well, this fw works on other chargers without mandatory issues. I'd start with your HW in the first place. Can you provide us more details? Schematics, your fw adaptation etc...

38293312, 2020-10-14:
> 好吧，这款固件可以在其他充电器上使用，而没有任何强制性问题。我首先要从您的硬件开始。您能否提供更多详细信息？原理图，您的fw适应等...

Thank you for your help . My hardware uses TBAMaxVer2.4, and the firmware uses TBAMAX-ver190_atmega32.hex

38293312, 2020-10-14:
[image]

closed 2020-10-14 by 38293312

diggit, 2020-10-14:
Now i see...
> I built a charger based on the hardware you provided. 

This repo home of cheali fw development. Hw you have is creation of independent person and is not directly affiliated with this repo.

38293312, 2020-10-15:
> 抱歉，该原理图的分辨率太低。Kicad添加到pdf的效果非常好...
> 
> > TBAMAX-ver190_atmega32.hex
> 
> 我在[十六进制目录中](https://github.com/stawel/cheali-charger/tree/master/hex/unstable)找不到这样的固件映像。
> 
> 首先，应该您[向该硬件的创建³³者寻求请立即获取iTunes](http://www.digiwood.ee/about-me)。看来提莫运作良好。
Thank you

38293312, 2020-10-15:
> > 抱歉，该原理图的分辨率太低。Kicad添加到pdf的效果非常好...
> > > TBAMAX-ver190_atmega32.hex
> > 
> > 
> > 我在[十六进制目录中](https://github.com/stawel/cheali-charger/tree/master/hex/unstable)找不到这样的固件映像。
> > 首先，应该您[向该该硬件的创建³³者寻求请立即获取iTunes](http://www.digiwood.ee/about-me)。看来提莫运作良好。
> > 谢谢

https://github.com/TBAMax/cheali-charger-TBAmax
This is the author's page. I have "Question" on his page, and I will connect to your page in the future, strange!