stawel/cheali-charger #301: Fixing incorrect voltage reading on discharge

y75ra, 2020-11-16:
So theres hardware-based problem with measuring discharge voltage. On discharge current flow across charge shunt resistor and charge transistor, but because current direction is inversed the voltage is negative and can't be measured by microcontroller. So voltage in discharge mode is incorrect as well as internal resistance.
So to fix this i tried to make active rectifier on OpAmp that do Vout = |Vin| so i can invert negative voltage and measure it.
As far as i know in charge mode it measure (BAT+) and (BAT-) so Vbat = (BAT+) - (BAT-)
But i faced problem that in discharge mode it seems to not measure (BAT-) at all. So i would like to know if its possible to change firmware to perform same Vbat measurement as in charge mode.

In case anyone interested in active rectifier i'll attach KiCad files, but without firmware modifcation theres no real use of it.
[Full-wave active rectifier.zip](https://github.com/stawel/cheali-charger/files/5546162/Full-wave.active.rectifier.zip)

diggit, 2020-11-20:
Well, you could have tell us what was wrong. Maybe some other people would benefit from it in future...

bra1nslayer, 2023-01-08:
@YuBeDev 

I have same issue as yours.How could you managed to solve this?Charging works fine but discharging voltage readings are not correct while discharging.I calibrated device many times still have issue.I also modified my charger to discharge at 5 amps instead of 1 amp.

YuBeDev, 2023-01-08:
Well, I didn't managed to solve this at that time and stopped spending more effort on it. For me its not that critical because my IMAX isn't modified and at 1A or so that measure error isn't critical. For some harder loads I bought 20A electronic load that can measure mAh as well.

bra1nslayer, 2023-01-08:
@YuBeDev 
@diggit 
Thanks for reply.

Let me tell you what i have modified.

I have replaced discharging resistor with two 0.22 ohm and 0.15 ohm resistors **in parallel** which makes it about 0.09ohm resistor for R50.This resistor change is made for high discharge current measuremens.
[image]

Replaced IRFZ44N with 4 mosfets to distrubute load and heat and mounted them on big heatsink with fan.

[image]
[image]
[image]

Compiled firmware for 5000ma discharge and modified discharge power to 25w.

When discharging about 1.5 amps voltage difference with b6 and fluke meter is about 0.1v but when i increase current about 5 amps voltage difference is about 0.4 volts and charger says discharge completed in about 30 seconds.Also noticed when draining high currents b6 voltage drops constantly and eventually it comes to 3v and charger says discharge completed.

Now i am suspecting about LM2904.Maybe this opamp is giving bad results to microcontroller?Can i use your circuit to replace LM2904?
What about R14 resistor?Should i modify it also?

I am little bit confused what is causing problems.

As i said charging works normally and voltage-ampere calibrations went smooth.

Any ideas are welcome.

YuBeDev, 2023-01-08:
@bra1nslayer 
Well, I haven't done any modifications so can't be sure whats wrong. I think its might be problem with opamp gain (R14 you mentioned) but without such circuit and measuring I can't be sure. 
My circuit can be used just to invert negative voltage to positive so microcontroller can measure it, but in the end my attempt failed. 
There's might be some other limitations and additional checks done by firmware that causes such behavior

IMHO for such loads I advice to get something like [https://www.aliexpress.com/item/1005001345772275.html?spm=a2g0o.order_list.order_list_main.46.21ef1802z19Oqt](this) or [https://www.aliexpress.com/item/1005005024604305.html?spm=a2g0o.order_list.order_list_main.118.21ef1802z19Oqt](this) with lesser accuracy

bra1nslayer, 2023-01-08:
I see.I will change resistor and post results.

Investing to your electronic load might be smart move.

bra1nslayer, 2023-01-11:
@YuBeDev 
@diggit 

Ok i think i found the problem and solution. Maybe this helps you also.
The solution is simple. We will measure voltage from 3rd channel. Connect battery between BAL2 and BAL3 and voltage problem will be fixed.Firmware allows measuring voltage from any cell.

I think it's a design issue because BAL0 and BAL1 pins doesn't have any OP amps. I didn't understand why no opamps are used for BAL0 and BAL1.(maybe cost reduction?)

[image]
 

While browsing code **HardwareConfigGeneric.h** i noticed 

```
#define ENABLE_SIMPLIFIED_VB0_VB2_CIRCUIT
//TODO: should be implemented!
//when the imaxB6 discharges, the voltage on Vb0_pin
//(Vb0_pin = VBATT- on the balance port)
//drops below 0V so the ADC doesn't see it.
//This is why we see a bigger Vb1 resistance.
```

This note which i assume developers already know this issue and couldn't fix it without modding hardware.

R14 resistor is irrelevant in my case.So i left it as is.

I have changed AO4468 charge mosfet with various mosfets because Maximum Body-Diode Continuous Current is for this mosfet is 4 amps and i am draining 5 amps.Still testing this mosfet.Will share mosfet model when testing is complete.

[image]

Also in settings activate advanced menu and in battery settings check if **adaptive discharging** enabled. If you don't activate this option battery voltage will go down and discharging will end prematurely.

Will post updates if any.