stawel/cheali-charger #306: Imax clone 80W up to 200W.

ElitePlusPlus, 2021-07-23:
Hi! I have upgrade hardware, but can't up charge/discharge current. I try all firmware, but only one  is works correctly. I change HardwareConfig and now can set change current up to 15A on 2V (I need to charge NiCd cell 15-110Ah), but when correct error I charge 14/I discharge 14 max current must be set up to 4,5A... Original firmware allowed set current up to 6A... Its good, but too low for my needs. What I do wrong and can I do something?

diggit, 2021-08-08:
Hi, can you share, what kind of mods you did?
Error 14 means your limit is our of measurable range IIRC.

ElitePlusPlus, 2021-08-08:
Hi! THX a lot for answer!
I change all transistor and diodes to more powerfull - IRF4905, IRF3205 and MBR20100. Then I made new coil and change resistor in the charge and discharge circuits for 5W once. I use two resistors 0,05 Om in parallel for current and two 0.1 Om in discharge. And now I can charge up to 10A, but reading device is less than twice. I want to add two resistors more and reach 20 A charging current...
[image]
[image]
[image]
[image]

ElitePlusPlus, 2021-08-08:
And charging process...
[image]
[image]

diggit, 2021-08-09:
So generally your Rcharge shunt is half of default value which should double Icharge range (good).
I am  not entirely sure about difference between `Ismps` and `IsmpsSet` here https://github.com/stawel/cheali-charger/blob/55aca4c162b46b770a4d29e8c0beb1c0f934f5a4/src/core/calibration/Calibration.cpp#L115
but `IsmpsSet` check fails.
I guess, that your charger can't set output so high, that it would reach desired output current.

Can you share your Icharge calibration values? (1 pair per calibration point)

ElitePlusPlus, 2021-08-12:
1500 mA (3000 mA really) - 17646
300 mA (600 mA) - 1987

diggit, 2021-08-17:
>1500 mA (**3000 mA really**) - 17646

What do you mean?
If calibration point is 1.5 A, you have to adjust your calibration so you get 1.5 A on output and not 3 A.

ElitePlusPlus, 2021-08-17:
Yes, when I calibrate charger show 1500mA, then ammetr show 3000mA. I have resistor smaller then original - 0.025 Om.

ElitePlusPlus, 2021-08-19:
Ok! I recalibrate to 1500 - 1500. And max charge current reach 15A. But discharge current max 3800mA on 1,2V... I will try to change resistor to 2x0,05 Om.

ElitePlusPlus, 2021-08-21:
Ok!!! I made changes that I planed! Now I have near 6300 discharge current, but cell was not fully charged. Lets charge it.
[image]
[image]

ElitePlusPlus, 2021-08-21:
And now I want to change firmware for chardge current up to 30A.

ElitePlusPlus, 2021-08-21:
Big cells are waiting!!!
[image]

ElitePlusPlus, 2021-08-22:
Can anybody make some changes to firmvare? Need to add tripple charging algorithm for Ni-Cd cells...

diggit, 2021-08-22:
What do you mean by "tripple charging algorithm"?
Also flash of Atmega32 if filled almost to the brim.

ElitePlusPlus, 2021-08-22:
It is second stage of charge when after chardging 1C add 1/10C for few hours.
Maybe cut Pb charge?

ElitePlusPlus, 2021-12-04:
Hi all! I made a new one of power PCB.
[image]
[image]
 And I hope to reach dis/charging current up to 30A. Lets try...

JFingerle, 2023-09-07:
Offtopic: Really impressive hardware mod!