stawel/cheali-charger #310: Feature request: preserve calibration when updating firmware

TBAMax, 2022-01-03:
For development purposes it would be beneficial to be able to update flash firmware without losing previous calibration data in EEPROM. Currently there is some version check that always render previous calibration data unusable.

stawel, 2022-01-04:
> Currently there is some version check

I'm guessing you mean this:
https://github.com/stawel/cheali-charger/blob/master/src/core/eeprom.cpp#L56-L58

>  that **always** render previous calibration data unusable

that's not true, the code only resets your calibration if:
1. you have changed CHEALI_CHARGER_EEPROM_CALIBRATION_VERSION
  in https://github.com/stawel/cheali-charger/blob/master/CMakeLists.txt#L23
(or)
2. the eeprom was somehow corrupted (crc check)
(or)
3. you just erased the eeprom during flashing.

I'm guessing in your case it's 3.,
maybe this will be helpful:
https://electronics.stackexchange.com/questions/98891/is-there-a-way-to-preserve-eeprom-contents-in-avr-atmega-when-burning-a-new-firm

TBAMax, 2022-01-04:
@stawel thanks. I went through the checklist and it was indeed
3. I used `avrdude.exe -patmega32 -cusbasp -Uflash:w:firmware.hex:i` and it turns out it clears the EEPROM also by default. That possibility did not occur to me.
 
The fix was just to change the fuses and burn with the following command instead:
`avrdude.exe -patmega32 -cusbasp -Ulfuse:w:0xAF:m -Uhfuse:w:0xC1:m -Uflash:w:firmware.hex:i`

This is setting the fuse EESAVE=1 before doing the flash. And so prevents the EEPROM being cleared.
[https://www.engbedded.com/fusecalc/]

I do not know what fuses are everybody else using. Is there a reccomended setting? I will run this for some time, maybe we can add the recommended setting if there is not one yet.

closed 2022-05-21 by TBAMax