stawel/cheali-charger #312: flash well done but not working (Need Help)

adop-adop, 2022-01-07:
Hello 
I need help with the programing my H6016 charger with ATmega32 µC
I have Flashing the .hex file " cheali-charger-Turnigy-Accucel-6-50W-5A_2.00-e10.3.12-20160613_atmega32.hex "

but after the flash nothing work any more 
my charger is the same of the charger flashed  is this post https://groups.google.com/g/cheali-charger/c/5JTsKRPkL2U

Ps: I have make a backup before flashing the cheali-charger firmware using the cmd :
avrdude.exe -patmega32 -cusbasp -Uflash:r:flash.bin:r -Ulfuse:r:lfuse.bin:r -Uhfuse:r:hfuse.bin:r -Ueeprom:r:eeprom.bin:r 
avrdude.exe -patmega32 -cusbasp -Uflash:r:flash.hex:i -Ulfuse:r:lfuse.hex:i -Uhfuse:r:hfuse.hex:i -Ueeprom:r:eeprom.hex:i

and try to flash the backup eeprom and flash 
but not working the display stay with white and blank line and the buzzer make a little noise .

thanks

[image]
[image]

stawel, 2022-01-10:
try this:  https://github.com/stawel/cheali-charger#troubleshooting

adop-adop, 2022-01-12:
Hi thanks for your help
the response is under my eyes but I don't have see 
my charger work now!!! :-D

adop-adop, 2022-01-12:
Hi stawel

I have calibrate my charger Voltage and Current (Charge and discharge) Vin too, No problem with the calibration
I read and adjust the voltage and current correctly
I have create Lipo and L435 battery's according my battery's type and characteristic

but when I start charge or charge+balance, the battery not charge even after several hours.
the fan is turn on when I enter in charge or charge+balance I think is normal

the screen of my charger after several hours charging one battery LIHV 2S 4,35V 650mAh 
I precise that the battery is not full discharged and the 87% is the actual charge in the battery when I start the process.
for details "10,4V" blinking continually.

thanks 

[image]

adop-adop, 2022-01-13:
Hi good news my charger charge correctly

I have reset the setting and eeprom "full reset" and calibrate the voltage and current 
and now the output voltage is correctly read by the charger 

And I can Charge my battery !!! :-))

closed 2022-01-14 by stawel