stawel/cheali-charger #315: After some use the charger goes crazy

Denethor85, 2022-06-24:
Hello, I have a clone 80w charger purchased at Banggood.  I flashed the latest available firmware version.  When I turn on the charger and for the first minutes / hour of use everything is always fine, then suddenly the charger starts showing squares and strange symbols on the display and also stops responding to commands.  Sometimes I push a few buttons at random and after a while it starts working normally again for some time then it starts going crazy again.  What can it be?

Thanks

TBAMax, 2022-06-24:
@Denethor85, good that you bring this up. This is actually a common issue with those 16x2 LCD screens. I have seen this happen with other Arduino projects also. 
  Probable cause is some kind of static charge or supply noise that disturbs the screen and the screen do not recover from that.
  First remedy for this is to try another power supply. Usually the issue goes away.
 Long term, it is probably possible to add periodic LCD resetting with the code, so that when the screen goes blank at some point, it will at least recover after a while. Downside to that is that reset creates a possibly disturbing LCD blink, since it is not possible to detect the LCD malfunction and do a conditional reset (or maybe it is by reading back the LCD settings memory, I do not know, this would be have to be tested, but is hard to intentionally generate the given malfunction.).

xgg69, 2023-02-13:
yes @TBAMax, i've had this issue with lcd screens (not only 16x2) in arduino projects too so this is pretty common ...i guess